MichaIng/DietPi #10: DietPi banner clear screen

d-faure, 2015-05-17:
The dietpi-banner is clearing the screen, making the startup sequence hard to debug

Fourdee, 2015-05-18:
Hi d-faure,

I'll look into this. We may be able to add a debug flag in the /boot/dietpi.txt settings file to initially skip screen clearing.

Will get back to you.

d-faure, 2015-05-18:
I temporary solved the issue commenting the "clear" line in the dietpi-banner script.
IMHO, there's no need to clear the screen, as you're either using some graphic environment, or nothing at all when running some kind of server (and in this case, getting all boot sequence output is really helpful to debug startup issues).

Fourdee, 2015-05-18:
Whilst i agree with you for debugging purposes, the original purpose of clear in dietpi-banner is to "clean" the screen for the user. 

Not all users of Linux are advanced or looking to visually debug their installation. So, having a screen full of text does nothing but confuse the end user.

Users looking to debug can will use logs in /var/log.

dietpi-banner is also called from most of the dietpi scripts (where clear is needed).

I did look into making a "noscreenclear" flag in dietpi.txt, however, this would require an additional filesystem read each time dietpi-banner is called. It would also require the user to know that such a flag exists. 

For the moment, clear is here to stay and no changes have been made. I will now close this issue, if you disagree, please reopen it.

closed 2015-05-18 by Fourdee