MichaIng/DietPi #9: ramlog fail to start

d-faure, 2015-05-17:
While booting a DietPi+WebServer (version 57), the apache2 server seems to be started before ramlog:

```
`[FAIL] Starting ramlog-tmpfs 2.0Error: /var/log is in use... failed!
The list of open files: (You need to close below daemons if you want to start/stop ramlog manually)

COMMAND  PID     USER   FD   TYPE DEVICE SIZE/OFF   NODE NAME
apache2 1688     root    2w   REG  179,2     1751 135108 /var/log/apache2/error.log
apache2 1688     root    6w   REG  179,2        0 135109 /var/log/apache2/other_vhosts_access.log
apache2 1688     root    7w   REG  179,2     1935 135107 /var/log/apache2/access.log
apache2 1746 www-data    2w   REG  179,2     1751 135108 /var/log/apache2/error.log
apache2 1746 www-data    6w   REG  179,2        0 135109 /var/log/apache2/other_vhosts_access.log
apache2 1746 www-data    7w   REG  179,2     1935 135107 /var/log/apache2/access.log

Test result: ramlog cannot be started or stopped at the moment.
```

Fourdee, 2015-05-18:
Hi d-faure,

Many thanks for the bug report. I'll look into this today.

Fourdee, 2015-05-18:
This will be resolved in v58.

In the mean time, you can apply this workaround:
update-rc.d apache2 disable

dietpi-services will then run the Apache2 service during the end of boot.

closed 2015-05-18 by Fourdee