MichaIng/DietPi #26: Install behind proxy

dimpfelmoser, 2015-06-30:
I am trying to install DietPi behind a proxy. So far I did the following:

1) copy image V70 to SD with dd
2) quit the installer after booting
3) set up proxy in /etc/environment, /etc/profile, and /etc/apt/apt.conf.d/10proxy, reboot

At this point, apt-get update is working from the command line. However, the installer hangs with the message "Waiting for apt-get update to finish: Please wait ..."

4) run dietpi-update to upgrade to V73

Still the same problem.

Regards, Marc

Fourdee, 2015-06-30:
Hi,

Thanks for the bug report.

The hang is caused because DietPi is waiting for NTPD to finish (so we get a updated timestamp), however, the onscreen text does not display this. 
A fix is available in the next release (v74). This will wait for a maxium of 8 seconds before killing NTPD. https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-apt-get_update#L102

dimpfelmoser, 2015-06-30:
Thank you. However, the state file ist left with state 0 after reboot, because I had to SIGTERM the apt-update after the DietPi-version-update, in order to reboot. If apt-update runs again, it waits for the update to terminate although it's not running, so it waits forever.

Even in my V73 the NTPD-query should timeout after 10 s. But it hangs forever.

In other words, Wait_For_Completion() waits forever, because Get_Current_State() always returns a status of 0.

I think the root of the evil is that apt-update is called at all after the DietPi-update (from V70 to V73 in my case), instead of just rebooting.

I could resolve the issue by deleting the state file.

Fourdee, 2015-06-30:
>> Even in my V73 the NTPD-query should timeout after 10 s. But it hangs forever.
DietPi runs ntpd with -gq flags (quit when done, to save memory). So unless it updates, NTPD wont terminate.

>> In other words, Wait_For_Completion() waits forever, because Get_Current_State() always 
>>returns a status of 0.
Yes, as per my previous comment, this is resolved in the v74 code i linked.

>>I think the root of the evil is that apt-update is called at all after the DietPi-update (from V70 to V73 
>>in my case), instead of just rebooting.
DietPi runs dietpi-apt-get_update in the background. This vastly reduces the waiting time during installation for 2> core devices.
It will also only run apt-get update every 7 days, avoiding the need to run apt-get update everytime the user installs software with dietpi-software.

closed 2015-06-30 by Fourdee

Fourdee, 2015-06-30:
I forgot to mention. You will need to use the v74 image release (released later today). The online patch will not effect v73<=.