MichaIng/DietPi #27: Network setup with German locale

dimpfelmoser, 2015-06-30:
If the locale is set to de-de-UTF8 in the config dialog prior to setting up a static network address, there are erroneous strings prepended to the ip address and mask, i.e. instead of

192.168.0.1

the installer writes

Adresse:192.168.0.1

into /etc/network/interfaces

Fourdee, 2015-06-30:
Hi,

Many thanks for the report. Could you please run:
dietpi-bugreport

This will generate a bug report with system configuration data. You dont need to fill the information in. Once its sent, please paste your unique ID code here. I'll then be able to look into this further.

Fourdee, 2015-07-02:
Unable to replicate on RPi + DietPi v74. 
Correct values are applied to /etc/network/interfaces when set to de-de-UTF8 before and after reboot.

closed 2015-07-02 by Fourdee