MichaIng/DietPi #32: Dietpi-update 404 error

ictinc, 2015-07-12:
Hi Fourdee,
Since about a week it seems not possible anymore to run the dietpi update script on my rpi2. 
I'm able to ping the host so it doesn't seem like a local network issue.

D I E T - P I | 05:43 | Sun 07/12/2015                                                                                               ───────────────────────────────────────                                                                                              V75 | Raspberry Pi 2 (1GB)                                                                                                           ───────────────────────────────────────                                                                                              DietPi-Update: Checking for DietPi updates, Please wait...                                                                                         --2015-07-12 05:43:02--  https://raw.githubusercontent.com/Fourdee/DietPi//dietpi/server_version                                     Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 23.235.43.133                                                     Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|23.235.43.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found                                                                                2015-07-12 05:43:06 ERROR 404: Not Found.                                                                                            root@rpi2:~#

Kind regards,

Ronald

Fourdee, 2015-07-21:
Hi ictinc,

Many thanks for the bug report. Apologies for the delay (was moving house).

Could you please run dietpi-bugreport and follow the onscreen instructions. This will help me debug the issue your experiencing.

Kind Regards,
Dan

ictinc, 2015-07-22:
Hi Dan,

Thanks for your reply, no worries for the delayed response.

I've sent you a bug report as requested which was filled under eference number;
b827eb55b259-0

Kind regards,

Ronald

Fourdee, 2015-07-22:
Hi Ronald,

Many thanks for the bug report. I've checked through your settings. It appears at some point, your device wasnt shut-down correctly (eg: power cut). Because of this, some of your settings have been corrupted.

A patch to prevent the above from occurring will be available in the next release (v78).

I would suggest a fresh image installation.

Kind Regards
Dan

Fourdee, 2015-07-22:
I would recommend a fresh image installation of DietPi where possible. However, you can manually patch your current installation. 

Running the below code will reset all DietPi specific settings back to default:

cd /DietPi
wget https://raw.githubusercontent.com/Fourdee/DietPi/master/boot.ini
wget https://raw.githubusercontent.com/Fourdee/DietPi/master/config.txt
wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi.txt

closed 2015-07-22 by Fourdee