MichaIng/DietPi #37: default root password on german keyboard...

jedie, 2015-07-30:
the default root password is "raspberry"... Nice. But on german keyboard with "english" layout you have to remember that "y" is under "z" key...

so, maybe it is a better idea to change the default to something else or set the keyboard layout before first login is needed...

Fourdee, 2015-07-30:
I'am not keen on changing the default password as this issue would still occur on the official Raspbian image regardless. 

Would a notification on the login screen assist?
eg:
default username=root
default password=raspberry
notes: German keyboard, password is raspberrz

jedie, 2015-07-30:
> Correct me if i'am wrong, but wouldnt this issue still occur on the official Raspbian image aswell?

I don't know if the default password is also **raspberry**
But IMHO in Raspian you end in the config tool on first boot, without the need for login. So i have first change language stuff and the default password...

> Would a notification on the login screen assist?
> ...
> notes: German keyboard, password is raspberrz

Yes, as a quick work-a-round. But this is only ok for german users and all others?!?

jedie, 2015-07-30:
What is the best solution here?

IMHO:

First question to the user (without login):
* used local and keyboard layout
* new root password

As for security: the root password should be changed on every system, isn't it?
But it's needed to set the right keyboard layout before: Otherwise the user set a password that will also not work with changes layout ;)

Fourdee, 2015-07-30:
> I don't know if the default password is also raspberry

On the official Raspbian, the user is pi, pw is raspberry.

> But IMHO in Raspian you end in the config tool on first boot, without the need for login. So i have first change language stuff and the default password...

I completely forgot about this. Your spot on. I'll see if i can implement auto login during 1st run into DietPi. It may have to wait for a week while i finish up other stuff, but i'll leave this open.

jedie, 2015-08-03:
What's about to change the default root pass to: **raspbian** ?

See: https://github.com/debian-pi/raspbian-ua-netinst#first-boot

Fourdee, 2015-08-03:
> What's about to change the default root pass to: raspbian ?
I'am tempted, for now, to go with root pass: dietpi . Would that work on a qwertz layout?

jedie, 2015-08-03:
:+1: 

> Would that work on a qwertz layout?

yes

Fourdee, 2015-08-04:
All DietPi images will use the new default password "dietpi"
https://github.com/Fourdee/DietPi/blob/testing/dietpi/boot#L260-L262

The new images (v81) will be available in the next few days.

closed 2015-08-04 by Fourdee