MichaIng/DietPi #38: install a little confusing...

jedie, 2015-07-30:
The first start (after some reboots) will end in:

* choose software you want to install
* dietpi-config

So i answer the questions like which software to install...
Then i use dietpi-config and configurate everything... But then it will reboot, so that changes are made...

But then i'm back on the same "select software" GUI... and my answers (made before) are gone and i must select my software again...

that is confusing ;)

Fourdee, 2015-07-30:
Hi Jedie,

Thanks for the feedback. I'll look into this and get back to you.

Fourdee, 2015-07-30:
DietPi-Config will no longer ask for a reboot during the 1st run stage of dietpi-software. https://github.com/Fourdee/DietPi/commit/9a1dbb5efcd30088fc8af2637711f690556bcc82

closed 2015-07-30 by Fourdee