MichaIng/DietPi #39: debian jessie...

jedie, 2015-07-30:
Is it's save to dist-upgrade to debian jessie?!?

And when will this project switched to jessie?

mamuesp, 2015-07-30:
Jens,

at first: I mentioned the same flaws as you did to Dan, he is aware, I'm sure he will adapt something.

The password "raspberry" is already the initial PW on Debian, so no shame on Dan - I'm sure one can live with that y <-> x swap.

The second flaw - configuration and installation and then, a big part of the configuration was ignored - I fell in too, but I helped myself - a second configuration run fixes all.

And yes, for my part I switched to Jessie with no problems - or at least none remarked ...

Regards, Manfred

Fourdee, 2015-07-30:
DietPi does not currently support Debian Jessie. There are a few reasons for this, mainly that most of the DietPi code would need to be updated to work with Jessie.

When the official Raspbian image updates to Jessie, i will then work on getting DietPi available for Jessie also. Until then, i wouldn't recommend using apt-get dist-upgrade.

jedie, 2015-07-30:
>  i wouldn't recommend using apt-get dist-upgrade.

good to known ;=)

roheve, 2015-09-26:
I did upgrade my RPI to jessie a while ago. Upgrading went well, but after the final reboot it got in to a recovery console mode. This was caused by two lines in fstab that try to mount /dev/sda1 to /mnt/usb_1 and I do not have an USB disk connected. I just commented out those lines and my RPi works fine now.
As I am not sure what will happen when I run dietpi-upgrade i did not do that (so I am stuck at v82), but running jessie and a newer nginx. :)

Fourdee, 2015-09-30:
> This was caused by two lines in fstab that try to mount /dev/sda1 to /mnt/usb_1 and I do not have an USB disk connected. I just commented out those lines and my RPi works fine now.

Hi Roheve, thanks for the information.
When I come to updating DietPi-RPi to Jessie, i'll be sure to check those fstab mounts for issues.

Fourdee, 2015-10-01:
@roheve 

Adding nofail to the /etc/fstab mount points prevents systemd from crying (and dying) when the device doesn't exist.

Fourdee, 2015-10-07:
DietPi for Raspberry Pi's (Jessie) released with v94:
Please see dietpi.net for download.

closed 2015-10-07 by Fourdee

dogmaticslumbers, 2015-12-23:
No Chromium in DietPi Jessie.  Really?

Fourdee, 2015-12-23:
@dogmaticslumbers 

Chromium for RPi:
http://conoroneill.net/running-the-latest-chromium-45-on-debian-jessie-on-your-raspberry-pi-2/

Chromium for Odroid:
apt-cache search chromium-

I'll look into the LXDE install and see if we can switch from Iceweasel to Chromium for all devices. As  the installation is different for some devices, each one must be tested.
https://github.com/Fourdee/DietPi/issues/150

dogmaticslumbers, 2015-12-23:
@Fourdee. Thanks!!  I knew there had to be a reason and that it had to be out there somewhere.

I'm off to get it installed now.

dogmaticslumbers, 2015-12-23:
OK.  Ended up following the instructions here:

https://www.raspberrypi.org/forums/viewtopic.php?uid=160930&f=63&t=121195

Even then, Chromium didn't fire up immediately after the install.  It was missing one dependency. (Sorry don't remember the name of the dependency.  Not at that board right now.  I followed the instructions for fixing the dependency by issuing some kind of apt-get command which caused the board to roll back then re-install a dozen or so files.  I'll repost here when I get back to the machine and look over the command history.)  The good news, Chromium is running!