MichaIng/DietPi #42: [ToDo] DietPi installation option for SubSonic

Fourdee, 2015-07-30:
http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=31&start=10#p169

Fourdee, 2015-08-05:
Notes to self:

SubSonic accepts startup parms in /etc/default/subsonic. However, I'am unable to find any reference of SUBSONIC_ARGS online. This prevents me from setting up the users media folder locations during DietPi installation.

Fourdee, 2015-08-06:
subsonic --help

thechildofroth, 2015-08-10:
Default folder locations are set by editing /usr/share/subsonic/subsonic.sh

Fourdee, 2015-08-11:
> Default folder locations are set by editing /usr/share/subsonic/subsonic.sh

Thanks i'll take a look. I'am currently using the subsonic command line options to set folder locations https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L2160-L2164

thechildofroth, 2015-08-11:
Ah ok, does this help:

To change the port number, Java memory settings or other startup parameters, edit the SUBSONIC_ARGS variable in /etc/default/subsonic. 

From
http://www.subsonic.org/pages/installation.jsp

thechildofroth, 2015-08-25:
Version 5.3 Beta2 running nicely on DietPi - had to rewrite symlinks to transcoder to complete upgrade:

ln -fs /usr/bin/ffmpeg /var/subsonic/transcode
ln -fs /usr/bin/lame /var/subsonic/transcode

http://www.subsonic.org/pages/download.jsp

Fourdee, 2015-08-25:
HI, thanks for the details.

Subsonic is already coded into dietpi-software, but is currently disabled as I havnt had the time to test it yet. I'll try and take another pass at it today. 

If you would like to test it, you can enable the installation option by:
- Uncomment the following line https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L3307
- Cut and paste the line into https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L3266 (just above)

thechildofroth, 2015-08-25:
Cool, thanks Dan.

I did wonder as I'd see you post it in the forum and then it disappeared!
I'll test it out tonight and let you know how I get on.

thechildofroth, 2015-08-26:
The basic installation of Subsonic works as expected but there's a bit of tidying to do I think.

We need to use Oracle Java rather than OpenJDK/IcedTea (which is painfully slow) + I think it reduces the number of packages that need downloading.

The script needs to install ffmpeg and lame too, and subsonic's transcoder needs redirecting to then using:

ln -fs /usr/bin/ffmpeg /var/subsonic/transcode
ln -fs /usr/bin/lame /var/subsonic/transcode

I'll do another test build tonight to remind myself what packages I installed in what order for my first working system as it seems to perform much better than this install (which is dropping out audio regularly).

Fourdee, 2015-08-26:
Great work on this!
As you mentioned, if you can list the packages required, i'll update dietpi-software.

thechildofroth, 2015-08-26:
Hi Dan,

I've just spun up a new image and gone through the whole process (which I've put here for completeness, obviously not all of it is relevant to the final script)

imaged 081
edited dietpi.txt for wireless

booted and left to resize disks etc

ssh'd in and allowed update patches to run + reboot

select to install alsa (not sure if this is necessary?)
selected hifiberry driver (obviously this is specific to my set up with an i2s soundcard

go start installation

The subsonic part.

apt-get install lame ffmpeg oracle-java7-jdk

at this point I did 'poweroff' and dropped the subsonic .deb in /mnt using my laptop and then reloaded the pi – equivilent to wget'ing the deb from sourceforge

(cd /mnt)
dpkg -i subsonic-5.3.beta2.deb

then redirect subsonic's transcoder using:
ln -fs /usr/bin/ffmpeg /var/subsonic/transcode
ln -fs /usr/bin/lame /var/subsonic/transcode

wait a while for subsonic to establish itself and away you go!

Fourdee, 2015-08-27:
Excellent job thechildofroth +1 . 
I've implemented your changes and subsonic works flawlessly.
It also seems to perform faster with oracle-java as you mentioned eariler. Great spot on the symlinks, i did wonder why .ogg wasnt playing the last time i tested it lol.

Again, great job and thanks for helping to make DietPi better and better! I've given you full credit in the changelog https://github.com/Fourdee/DietPi/blob/testing/CHANGELOG.txt#L5

We should be releasing v85 today (its a biggy) and this will enable the SubSonic installation option.

I best crack on with adding the forum page for SubSonic (again) ;) lol

Fourdee, 2015-08-27:
Released in v85

closed 2015-08-27 by Fourdee