MichaIng/DietPi #46: New logo and website

dyzajash, 2015-08-10:
Hello, 
congratulations for creating this awesome distro! I'm using it in my company with Rpi2. 

I want to help you with:
* Create a modern and clean website for DietPi
* Create a logo for DietPi.

So I have some questions: 

* Can I change a shape of logo? 
* I think silver and purple are great, but I want to make full scheme for better identity 
* Can I change typography?

Fourdee, 2015-08-10:
Hi Dyzajash,

Thanks for getting in touch and offering your skills to the project. Yes, we are in desperate need of a clean and modern website, aswell as a new logo for DietPi.

To answer your questions, you have free reign on everything. This should allow you to put your own twist on the designs and bring out your best work with your own touch.

If you need any information, and/or, wish to send me your prototypes, please get in touch.

Fourdee, 2015-09-21:
Inactive.

New website is live http://dietpi.net
Many thanks to Jerome Queneuder for providing the modified template and his assistance!

closed 2015-09-21 by Fourdee