MichaIng/DietPi #47: Enable UART login by default

JarrettR, 2015-08-11:
For headless and networkless setup. Connecting to the UART pins on the GPIO headers with a bridge allows you to read the boot log, but actually logging in and using it as a terminal requires some tweaking.

I believe I was able to do this in v61 by changing cmdline.txt to:
> dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 elevator=deadline root=/dev/mmcblk0p2 init=/bin/systemd rootfstype=ext4 rootwait

And uncommenting the last line in inittab.

This results in additional startup text that may not be desirable, so there might be a better way of doing this.

Additionally, I was not able to get this working with v81, so either something change, or I have forgotten a step.

Fourdee, 2015-08-12:
Hi Jarrett,

I have made no changes to UART that i'am aware of, however, as I have no experience with the above, it could of been caused accidental during patching.

I'll need to obtain the required hardware before I can be of any assistance. In the mean time, the only thing i can recommend is to try the following for cmdline.txt

> dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop rootwait

Also, which version of DietPi images are you using? Can you also remember which image version you had that worked previously if it was different?
To find out the current image version, simply write the image file, open the boot partition /boot/dietpi/.version

JarrettR, 2015-08-18:
I vaguely recall that init systemd was important.

Sorry, I'm not being helpful here. Set it up successfully in v61 a couple months ago, and in v80, I couldn't figure it out.

I'll go through the process again at some point soon and report back.

Fourdee, 2015-09-13:
No response. Please reopen ticket if required.

closed 2015-09-13 by Fourdee