MichaIng/DietPi #51: Services Fails after Reboot

Bobbaone, 2015-08-17:
Hello,

ive freshly installed dietpi - its amazing. But if i reboot, apache2, munin... and others arent starting.
i have manually start them. Is this coused by the fact, that the logs are at the ram?

Is it possible to set a "slowdown" for the services, after all system tasks has been loaded?

Greetings.

Fourdee, 2015-08-17:
Sounds like you'll need to check your init startup orders.
Check your /var/log/apache2 logs to see if it explains whats happened.

If you install software thats not from the dietpi-software system, you may need to manage your startup orders depending on how those startup scripts are coded.

You can check your startup order by running:
ls /etc/rc*.d/
Check and ensure dietpi-service starts before apache2 minun (00 being the first, 99 being the last).

You can check to see if the /var/log disk is mounted as TMPFS:
df -h | grep "/var/log"

Bobbaone, 2015-08-22:
This is, what i get:

http://prntscr.com/87nijs

Fourdee, 2015-08-22:
Apache 2 is running before the DietPi-Service. Did you install Apache2 from DietPi-Software, or, apt-get?

The DietPi-service sets up the ramdisk /var/log directory, and its subfolders from last shutdown. Basically, Apache2 is starting before /var/log/apache2/ can be generated. This will fail Apache2 starting up.

Use the following to resolve the issue:
> service apache2 disable

DietPi will then take over and start apache2 after the DietPi-Service has completed.
Alternatively, install the Apache2 webserver stacks available in dietpi-software.

Fourdee, 2015-08-22:
Issue resolved.

Please reopen if required.

closed 2015-08-22 by Fourdee