MichaIng/DietPi #52: Syntax error in dietpi-config

JarrettR, 2015-08-18:
To reproduce:

Connected via SSH, no connection to the outside world.

Hitting "Test network" in the network config menu results in this:

````bash
cat: /DietPi/dietpi/.installed: No such file or directory
/DietPi/dietpi/dietpi-config: line 1332: ((: == 2 : syntax error: operand expected (error token is "== 2 ")
cat: /DietPi/dietpi/.installed: No such file or directory
/DietPi/dietpi/dietpi-config: line 1352: ((: == 2 : syntax error: operand expected (error token is "== 2 ")
cat: /DietPi/dietpi/.installed: No such file or directory
/DietPi/dietpi/dietpi-config: line 1369: ((: == 2 : syntax error: operand expected (error token is "== 2 ")

Testing connection to http://mirrordirector.raspbian.org, please wait.....
````

Minor issue, doesn't seem to negatively impact anything.

Fourdee, 2015-08-18:
Hi Jarrett,

The only reason this will occur is if something went wrong with the .installed file. As this file is generated on 1st run, and before you can access dietpi-config, you should not see this issue.

Can you please replicate the error on your system, then, run dietpi-bugreport and provide me with the reference code. This will allow me to see your .installed file and debug.

JarrettR, 2015-08-18:
That might be another piece of the puzzle. This occurs during first-run, while it's trying to connect for updates for the first time. It's happened on two different reflashes.

If you still need the bugreport reference, I will grab that this evening.

Fourdee, 2015-08-19:
Hi Jarrent,

Thanks for the update.

>  This occurs during first-run, while it's trying to connect for updates for the first time. It's happened on two different reflashes.

Which image version are you currently using? eg: DietPi_v81_RPi.img

Fourdee, 2015-08-31:
No response. Please reopen ticket if required.

closed 2015-08-31 by Fourdee