MichaIng/DietPi #57: Add NGiNX and Redis installation option

dyzajash, 2015-08-20:
Please consider adding Nginx 1.8 [not LEMP] (latest stable) install option, and Redis 3.0.X (repo has old 2.4 option). 

Btw. I have few sketches of design, so next week I'll add them inside design issue.

Fourdee, 2016-03-09:
note to self:
non-SQL based data/key storage server.

blindpet, 2016-03-09:
Unfortunately the latest redis in repos is always ancient and to my knowledge they do not have an official repo. It is easy to script building from source. I use [these instructions](http://www.htpcguides.com/install-configure-latest-redis-cache-wordpress/)

Fourdee, 2016-03-09:
@blindpet

Thanks for the info. I'll give the compiling a spin.

notes to self: compile .deb for
ARMv6
ARMv7
x64

Fourdee, 2016-03-20:
v113 will contain the installation from apt. When I continue work on EmonPi https://github.com/Fourdee/DietPi/issues/210 I will modify the install to latest version and compile.

Fourdee, 2016-03-20:
Note to self: Add usage details to dietpi.com/software?

Fourdee, 2016-03-20:
Completed with apt-get for v113.

Fourdee, 2016-03-20:
Moved to v114 to improve it using latest redis server version, which will be from compiled source.

Fourdee, 2016-03-20:
Created a new ticket for v114 to update the current apt-get install to latest version (from source): https://github.com/Fourdee/DietPi/issues/244

closed 2016-03-20 by Fourdee