MichaIng/DietPi #64: Reboot causes two Subsonic processes to be spawned

thechildofroth, 2015-09-05:
When I reboot Dietpi it returns to two subsonic processes showing in htop (which causes jetty to fail - I get a 503 error when trying to connect to the web interface) seems I have to kill both of them and do 'service subsonic start' to get it going properly.

Also I think the minimum memory size you've allocated in subsonic_args is a little too small as I was getting audio dropouts even in Jukebox mode (with no transcoding) - I've increased mine to 350 and it's smooth now, that may be overkill but I would suggest bumping it back up to maybe 150?

Fourdee, 2015-09-05:
Hi,

> When I reboot Dietpi it returns to two subsonic processes showing in htop

I was unable to replicate this, could you send a screenshot, or "ps aux > ps.txt".

> I get a 503 error when trying to connect to the web interface)

There is a 60second delay from a reboot before Subsonic "fires up". The process is always there, but it appears Subsonic has an issue with loading and processing the rest of its program (eg: web interface).  Not sure what its waiting for, could be a retry attempt. This doesnt occur on a service restart.

> minimum memory size

This will be fixed in the next release. https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L2382-L2387

thechildofroth, 2015-09-05:
Here's ps aux from a fresh reboot: http://pastebin.com/bwwNXUuQ

I'd noted the delay on things getting going, normally while it's not fully fired I just get 'cannot connect' for the web interface, when the 2 threads are running I get a different error.
[image]

Fourdee, 2015-09-07:
> Here's ps aux from a fresh reboot: http://pastebin.com/bwwNXUuQ

Thanks for the requested information. I'll have a look at this over the next few days and report back when I have some results.
Regards,
Dan

Fourdee, 2015-09-13:
HI,

Apologies for the delay. I've been unable to replicate the 2 processes on my RPi. However, it may be caused by an issue with the pid file and subsonic service script.

Could you please try the following:
> update-rc.d -f subsonic disable

Then reboot the system. Let me know how many subsonic processes there are.

thechildofroth, 2015-09-22:
Hi Dan,

Just got back from a few days away so sorry for the delayed response. 

I updated to v90 and on reboot I still had two processes running so I did the update rc.d thing and that seems to have cleared it (just one process running on reboot), thanks.

closed 2015-09-22 by thechildofroth

Fourdee, 2015-09-22:
> Just got back from a few days away so sorry for the delayed response.
I updated to v90 and on reboot I still had two processes running so I did the update rc.d thing and that seems to have cleared it (just one process running on reboot), thanks.
No worries. Good to hear is been resolved!

I've changed the way DietPi controls services in v92 which will also resolve the issue.