MichaIng/DietPi #68: Kernel Panic

Bobbaone, 2015-09-16:
Since the last Update, suddenly i got a kernel panic.

I cant connect via SSH.
Via HDMI, there only Shows the Kernel Panic message...

I got a clone from the SDCARD. If you want, i can upload it to any Hoster.
All my Datas are lost, im little bit angry...

Greetings.

Fourdee, 2015-09-16:
Hi Bobbaone,

I very much doubt the update caused this issue. It is most likely caused by a failing flash media (eg: sd card), or, failing PSU.

Please try to answer these questions:
- What device are you using (eg: rpi)?
- What version was your DietPi system running, before the update (eg: v77)?
- What is the votage and amperage of your power supply?
- Are you over-clocking your system?

In regards to your data, if your using a USB HDD, your data will still be intact. However, if your using just a flash media card, you will need to plug it into a Linux system to read the 2nd partition (ext4).

Bobbaone, 2015-09-16:
i recently installed dietpi freshly.

Is it possible to get my Data from /etc/*?

Is it easy to get? Possible with Windows or Mac - i use both.?!

Thanks!

Fourdee, 2015-09-16:
Hi Bobbaone,

You will need to run a system that supports ext4 filesystem. Any Linux distro would work, a Debian live USB stick would be a good choice. See here for more info: http://www.pendrivelinux.com/debian-live-flash-drive-install-from-windows/

If you can tell me what programs you had installed (eg: bit torrent transmission, etc) I can tell you where the files/folders will be located.

Bobbaone, 2015-09-16:
oh, the program file are not needed - the only thing i need is the RRD Backup file from my weather station...

Fourdee, 2015-09-16:
This is new to me, so i have no idea where your backup would be stored. 
You are best to check the documentation of the software your using, for information relating to where the backups are stored.

Bobbaone, 2015-09-16:
i know where the files are but i dont come to...

this is the error with the live CD

Error mounting /dev/sdf2 at /media/user/1263ae8d-aaf3-41b6-9ac0-03e7fecb5d6a: Command-line `mount -t "ext4" -o "uhelper=udisks2,nodev,nosuid" "/dev/sdf2" "/media/user/1263ae8d-aaf3-41b6-9ac0-03e7fecb5d6a"' exited with non-zero exit status 32: mount: wrong fs type, bad option, bad superblock on /dev/sdf2,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.

Fourdee, 2015-09-16:
fsck /dev/sdf2

Bobbaone, 2015-09-17:
Its ok - ive settled with it..

closed 2015-09-17 by Bobbaone