MichaIng/DietPi #72: Add IPv6 support

Fourdee, 2015-09-21:


Fourdee, 2015-09-27:
I've planned to add a "toggle" in DietPi-Config, between ipv4 and ipv6. 

In the mean time...
IPv6 can be enabled in all DietPi images, by running the following code:
> rm /etc/modprobe.d/blacklist-ipv6.conf &> /dev/null
sed -i "/net.ipv6.conf.all.disable_ipv6 /c\net.ipv6.conf.all.disable_ipv6 = 0" /etc/sysctl.conf
sed -i "/net.ipv6.conf.default.disable_ipv6 /c\net.ipv6.conf.default.disable_ipv6 = 0" /etc/sysctl.conf
sed -i "/net.ipv6.conf.lo.disable_ipv6 /c\net.ipv6.conf.lo.disable_ipv6 = 0" /etc/sysctl.conf

Fourdee, 2015-10-20:
Does anyone have a IPv6 local network and willing to run some tests for DietPi?

audy, 2015-12-27:
Worked for me using DCHP (didn't test static).

```bash
$ uname -a
Linux DietPi 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015 armv6l GNU/Linux
```

MuJJus, 2016-01-15:
worked for me

Fourdee, 2016-01-15:
@audy @MuJJus 
Thanks guys.

I will begin to integrate this into DietPi-Config as a toggle option.

Fourdee, 2016-01-17:
Completed. Integrated into DietPi-Config as a toggle option. Available in next release (v107):
[image]

MuJJus, 2016-01-18:
Want DHCPv6 support
`iface eth0 inet6 dhcp`
in /etc/network/interfaces

EUI-64 address is too long and no privacy

Fourdee, 2016-01-20:
IPv6 is now enabled by default on all DietPi systems, see link for explanation:
https://github.com/Fourdee/DietPi/issues/175

Fourdee, 2016-01-20:
@MuJJus 
```
Want DHCPv6 support
iface eth0 inet6 dhcp
in /etc/network/interfaces

EUI-64 address is too long and no privacy
```
Unfortunately, I do not have access to any IPv6 networks that would allow me to test and implement this correctly into DietPi.
Your request will have to wait until the above is available.

closed 2016-01-20 by Fourdee