MichaIng/DietPi #73: DietPi-Config | Add CPU/ARM min frequencies for RPi

Fourdee, 2015-09-23:
> https://www.raspberrypi.org/forums/viewtopic.php?p=819840#p819840

Fourdee, 2015-09-27:
Done as a toggle option. v93.

DietPi-Config (RPi) -> Performance Options -> Lower Min Freqs
New menu option to toggle the use of lower minimum freqs in config.txt(arm_freq_min=100 etc). Allows for potential power saving.

Fourdee, 2015-09-29:
Update, some issues when  using the _freq_min settings:
- Using any of the _freq_min settings causes initial_turbo to be disabled.
- Using a value of core_freq_min=75. Causes the display to turn on and off. A direct link to what state the scaling frequency is at that time. Off when CPU is idle, On when CPU is scaled to max.

Fourdee, 2015-10-06:
> https://www.raspberrypi.org/forums/viewtopic.php?p=822170#p822170

No response from RPi community, and, no official documentation which covers a compatible list of frequencies.

For the mean time, this feature will remain disabled until further testing, and/or, documentation is available.

Fourdee, 2016-05-17:
Marking this as closed. Min frequency feature will stay disabled for foreseeable future.

closed 2016-05-17 by Fourdee

MichaIng, 2019-10-10:
Works well now:
```
2019-10-10 18:52:03 root@micha:~# grep -E '(arm_freq|initial)' /DietPi/config.txt
initial_turbo=20
#arm_freq=900
arm_freq_min=150
```
Lowest supported value is 100 AFAIK, although I can't find the source anymore. However it simply does not make any sense to set it lower, since ondemand governor would then nearly never settle down due to high % CPU usage. Note that RPi only jumps between lowest and highest freq but does not support multiple freq steps by default. So a too low minimum means maximum clocks run more often. Best tradeoff on my particular case was 150.

MichaIng, 2019-10-10:
Re-added: https://github.com/MichaIng/DietPi/commit/1082b1864dc07e665a2c3915f4d4cfb817910002