MichaIng/DietPi #78: DietPi-Software | DHCP server installation option.

Fourdee, 2015-10-03:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=102#p365

xenfomation, 2016-05-13:
Adding to my pile as it may be a good opportunity to roll into prototype for a DietPi database, the optimized menu ticket, and recent topics regarding layout changes.

xenfomation, 2016-05-14:
Update to @Fourdee 

### DHCPD SERVER
Adding the DHCPD server is not an issue and is done.

Before I commit this, I would like to offer an optional web-based config tool, such as IPPlan (aka iptrack), etc, but the problem is that many of these once popular tools are... stable, but really old.

So, before I assume, I wanted to ask @Fourdee if it would be okay to actually to add "DHCPD Server" and "DHCPD Web Config Tool" as options under Linux Software.

rhkean, 2016-05-15:
we already have a dhcpd server with pihole/dnsmasq

do we really want to add support for another one?

xenfomation, 2016-05-15:
Ha!  NO.  I did not know that and PLEASE FEEL FREE TO CLOSE!!!

Fourdee, 2016-05-15:
WiFi Hotspot installation uses DHCP server ```isc-dhcp-server```, so that could always be split and modified.
Although, kind of agree with @rhkean. Most of our users are home users, with a router and no specific need for just a DHCP server. Please correct me if i'am wrong :)

I'll mark this as closed. However, if people want to see a separated DHCP server installation option, post your interest here and we will monitor it.

closed 2016-05-15 by Fourdee

xenfomation, 2016-05-15:
Agreed as I am all about code, but not reinventing the wheel.  Network services are a must, but since we have it ... no need to break out a whiptail menu and offer a separate webstack to manage it (as of now! )