MichaIng/DietPi #79: /DietPi/config.txt does not contain the actual config.txt, despite what it claims

husnoo, 2015-10-04:
$ cat /DietPi/config.txt 
This file has been moved to DietPi-Ramdisk.
New location = /DietPi/config.txt

Fourdee, 2015-10-04:
Hi Husnoo.

What version of DietPi are you currently running? Are you able to send a dietpi-bugreport?

husnoo, 2015-10-04:
nawal@poobum:~$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
NAME="Raspbian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=raspbian
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

But report: 801f029201f3-0

Fourdee, 2015-10-04:
This is a known issue with v77. 

Please follow this post to resolve the issue:
http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=39&p=131&hilit=404#p131

Fourdee, 2015-10-06:
Assumed resolved as it is a known issue and resolution. Please reopen ticket if required.

closed 2015-10-06 by Fourdee