MichaIng/DietPi #84: Jessie breaks Subsonic

thechildofroth, 2015-10-10:
Just testing the new Jessie image, it seems that upstream have removed ffmpeg (replaced with libav - which doesn't seem to be a 'diet' solutions...)

Fourdee, 2015-10-10:
Thanks for the report. I'll take a look.

Fourdee, 2015-10-11:
I've updated the installation to install libav-tools. Subsonic runs but no sound, probably due to the old symlink we used as ffmpeg no longer exists in Jessie:

> ln -fs /usr/bin/ffmpeg /var/subsonic/transcode

Any ideas?

thechildofroth, 2015-10-11:
On the bit of reading I've done today it looks like Debian are switching back to ffmpeg in the next version (?!)

I'm just about to test installing ffmpeg from the debian multimedia repo (as per these instructions: https://www.assetbank.co.uk/support/documentation/install/ffmpeg-debian-squeeze/ffmpeg-debian-jessie/) and see if I can get it working that way in dietpi. What would the deal be about adding more sources to dietpi?

Fourdee, 2015-10-11:
Yeah i heard that aswell with the next Debian. Ideally, we need Wheezy-backports but its not supported by Raspbian.

I'll take a look at your link tomorrow and see what I can sort out.

thechildofroth, 2015-10-13:
I've submitted a but report as I'm somewhere close to the edge of my skills
in working out what's wrong.

From a fresh image I added the Debian Multimedia repositories before using
dietpi-software to add subsonic (which all seemed to work including pulling
in ffmpeg and LAME) but I've still got no sound and can't see ffmpeg
running in HTOP.

Fourdee, 2015-10-14:
Hi Marc?
I received your bug report, thanks for sending.

Ok, good news. This will be resolved in DietPi v97 :)

FFmpeg doesnt create a process, so it will never be seen in htop. Appears Subsonic handles this within its own process.

####The issue is caused by two things:
1 = An issue with creating 2 symlinks to the same source https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L2423-L2424 (how I missed this i'll never know)
2 = ffmpeg not available as a package.

You were spot on with your tests. We need to install ffmpeg using the Debian multimedia repo you linked, however, we do not need to build it from source.

####Heres what I did to resolve the issue on the current (v96) version of DietPi Jessie:
- install subsonic using dietpi-software
- Run the following commands:
> echo -e "deb http://www.deb-multimedia.org jessie main non-free" >> /etc/apt/sources.list
/DietPi/dietpi/dietpi-apt-get_update 2
apt-get install deb-multimedia-keyring -y --force-yes
/DietPi/dietpi/dietpi-apt-get_update 2
apt-get install ffmpeg -y
ln -fs /usr/bin/ffmpeg /var/subsonic/transcode
service subsonic restart

thechildofroth, 2015-10-15:
Thanks Dan,

I'll test this recipe, it's odd as I can't see what's different from what
I've done - but hey if it works now who cares!

thechildofroth, 2015-10-15:
I've flashed a new image, let it update and followed your instructions but
I'm still not getting any sound (either transcoded over the network or via
'Jukebox' mode).

(I've also used dp-config to reset to my dac soundcard)

Subsonic seems to think it's working, this is in the log when I play a file:

[10/15/15 12:51:19 PM BST]INFOPlayQueueInputStreamadmin listening to "Adele
- 21/01. Adele - Rolling in the Deep.flac"
[10/15/15 12:51:19 PM BST]INFOTranscodeInputStreamStarting transcoder:
[/var/subsonic/transcode/ffmpeg] [-i] [/mnt/usb_1/Adele/Adele - 21/01.
Adele - Rolling in the Deep.flac] [-map] [0:0] [-b:a] [192k] [-v] [0] [-f]
[mp3] [-]

In the working (Wheezy) version, when I play tracks in subsonic I get an
extra thread in htop showing ffmpeg doing its thing.

Fourdee, 2015-10-15:
Strange,

I did fresh installs on both Wheezy and Jessie. It appears the ffmpeg transcode process loads on both distros, however, once the whole file has been transcoded (which is quicker than playback), the process is stopped.

I also tested .flac files on both distros, worked fine:
[image]

I'am aiming to roll out v97 today with the updated SubSonic install. When its released, I would recommended writing a new Jessie image. I'll also post here when v97 is out.

Fourdee, 2015-10-15:
v97 released and confirmed working. Please reopen if required.

thechildofroth, 2015-10-16:
Sorry, but still no dice here. 

Reformatted, re-imaged and reinstalled - all seems to go well but I'm not getting the second process with ffmpeg and if I issue any command to ffmpeg from the prompt I get 'illegal instructions'

Fourdee, 2015-10-16:
Ah, sounds like your running the RPi v1 (armv6)? I did the Jessie tests on my RPi v2 (armv7)

thechildofroth, 2015-10-16:
Ok, that makes sense - probably worth putting a health warning on the Subsonic installation option.

When I get a chance I'll have a play at compiling - if I get a simple(ish) working recipe I'll post back as it might be worth including so that the installation is foolproof regardless of hardware platform?

Fourdee, 2015-10-16:
Yep by all means, if you get the compile working for your RPi v1, let me know the steps. I can then add it to the RPi v1 installation code for dietpi-software using Jessie.

thechildofroth, 2015-10-19:
Making the compilation of ffmpeg takes ages, is there a way I can share a compiled binary which dietpi can pull on if installing for this hardware version?

blindpet, 2015-10-19:
@thechildofroth ffmeg has rules about not posting precompiled binaries unfortunately

thechildofroth, 2015-10-19:
@blindpet thanks for the info, it doesn't look impossible from here: https://www.ffmpeg.org/legal.html but I am probably misunderstanding it, certainly 18 doesn't make sense to me at the moment.

Fourdee, 2015-10-19:
@thechildofroth Please send me the binary .deb package. 
I'll look into their "license claims" as it seems they are "picking and choosing" multiple licenses, but including/excluding "what they like".

At the end of the day, if they want to sue a GPLv2 project that makes no money, go for it.

thechildofroth, 2015-10-23:
Thanks Dan,

I'll do this but it will take me a while as you're pushing my debian chops
pretty hard (and the first compile didn't work).

I've not got as far as working out how to use a toolchain so we'll just
have to wait for the pi to crunch the numbers.

thechildofroth, 2015-10-24:
I mostly used this guide:

https://www.raspberrypi.org/forums/viewtopic.php?t=52834&p=406109

(when I configured it I also added --arch=armhf)

and came up with these, which work (woo)
https://drive.google.com/folderview?id=0B-Y8Pjy36wUZcEpxX2tnc2hVenM&usp=sharing

FFmpeg installs to /usr/local/bin/ffmpeg (so the location, or the symlink, needs changing when dp configures the subsonic install)

Fourdee, 2015-10-24:
@thechildofroth Lol, i'am just compiling ffmpeg for armv6 at the moment. You beat me too it :)
Once this compile is done, i'll test both .deb packages.

thechildofroth, 2015-10-24:
Neat, I've also compiled the fdk version of the aac codec, which saves having the MM repo in there and is apparently the superior audio codec for that format.

Fourdee, 2015-10-26:
Having a nightmare with this.

Using either your compiled ffmpeg.deb or my ffmpeg.deb, reports libvorbis missing. 
The only way to install this package is through the debian multimedia repo:
> echo -e "deb http://www.deb-multimedia.org jessie main non-free" >> /etc/apt/sources.list
/DietPi/dietpi/dietpi-apt-get_update 2
apt-get install deb-multimedia-keyring -y --force-yes
/DietPi/dietpi/dietpi-apt-get_update 2

However, from today, the ffmpeg package is now available in the deb multimedia repo, for armv6 and is considered an update to our packages.
If we install the ffmpeg from the repo, running ffmpeg results:
>root@dietpi-1:~# /usr/local/bin/ffmpeg
Illegal instruction

It appears one of the libraries that ffmpeg installs with the deb multimedia repo is broken.

So, we somehow need to include all the shared libraries during our compile. As my linux compiling knowledge is limited, i'll have to do some indepth research.
Or, wait until the debian multimedia repo resolves the broken packages.

thechildofroth, 2015-10-26:
Thanks for the update Dan,

I'll try a clean install with my .deb files - not sure about that
dependency (libvorbis) as I've not enabled any other repos to get it
working on my machine - did you install the AAC codec I compiled too?

It'll be worth getting this working as it'll likely also have an effect on
Ampache (as I think that uses FFmpeg too?)

Fourdee, 2015-10-26:
I didnt try the fdk originally, so ran a new test on fresh install. Same issue. http://pastebin.com/gyHhN1pG

Yes, ffmpeg is a shared resource and is required for Ampache, hence why i've spend so much time on it. Its just the RPiv1 (armv6) + Jessie left to do. XU4/C1 are compiled during install (takes <2mins), RPiv2(armv7) is available in debian multimedia repo.

thechildofroth, 2015-10-26:
I have FFmpeg working from a fresh install.

Need to install: libass-dev libopencore-amrnb-dev libopencore-amrwb-dev
libspeex-dev libtheora-dev libtool libvorbis-dev lame

I suspect it's not the dev file, but something it brings in with it, that
makes it work? So there's probably room for some efficiency in package
choice, but it doesn't need the other repo.

I'm going to get on with installing java and subsonic to give it a whirl

thechildofroth, 2015-10-27:
Confirmed working nicely with my compiled FFmpeg and FDK codec, running on
Java 8 and with the (released yesterday) new 5.3 version of Subsonic.

For additional 'dietage' it would appear that it is not necessary to
install alsa for subsonic to work.

[image: Inline images 2]

Fourdee, 2015-10-27:
@thechildofroth Doesnt Subsonic have the ability to play to local output? This would require alsa-base.

Thanks for the info on the -dev packages :+1: . I'll try it today and update installation code if all works. 
I'd like to try and find a method of installing without relying on -dev packages, seems a bit of a hack, however, if the -dev packages is the only option, we can use it for now.

I'll let you know results. Thanks for your assistance on this.

thechildofroth, 2015-10-27:
> Thanks for the info on the -dev packages [image: :+1:] . I'll try it

re; ALSA: Subsonic does play locally (and I've tested it as working) I'll
check later if ALSA base has been brought in without my noticing.

I can also try uninstalling the -dev packages and see if it keeps working,
is there a way I can use either the bug report or the backup tool to show
you what's installed?

Fourdee, 2015-10-27:
Yep, if you run dietpi-bugreport , it sends me list of installed packages (dpkg -l) , and, various other system and DietPi settings.

Fourdee, 2015-10-27:
Ok, tested all DietPi systems for FFmpeg and Subsonic. All working ok. 
- The -dev packages did the trick for armv6.
- Subsonic installed version is now 5.3 (not beta2)
- Java installed version installed is 8 (not 7)

Fourdee, 2015-10-27:
Done. Closing ticket.
Resolved in v99 which will be released later this week.

Reopen if required.

Mootch, 2015-10-27:
Will this update only install ffmpeg during the subsonic installation? I was looking to get ffmpeg working with ampache on a rPi2 with Jessie and wasn't sure if this fix would help for other applications.

Fourdee, 2015-10-27:
@Mootch 
DietPi optimized installation for Ampache will be available in v99 also (its already coded).

I've separated FFmpeg into its own installation option in dietpi-software > linux. All dietpi-software that requires FFmpeg will automatically be installed (which includes Subsonic / Ampache).

Mootch, 2015-10-27:
That's even better than I expected. I appreciate the work you are doing on this.

Fourdee, 2015-10-27:
@Mootch 
Thanks, appreciate it :)

I cant take all the credit for Ampache. "Jig" helped provide the installation steps for it: http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=146#p551

thechildofroth, 2015-10-27:
Great work Fourdee!

I've just been having a play with trying to whittle down the package list,
I did apt-get remove on all the -dev files, which, as you'd expect, broke
FFmpeg. However I've got it back up by installing the following packages:

libogg0
libvorbisenc2
libtheora0
libspeex1
libopencore-amrwb0
libopencore-amrnb0
libass5

So we can put this FFmpeg install on the DietPi Diet!

Fourdee, 2015-10-27:
@thechildofroth 
Brilliant, i'll update the installation code for v100. We like minimal stuff here :D

Fourdee, 2015-10-30:
Update installation code for v100.

Replaced -dev packages with:
apt-get install libogg0 libvorbisenc2 libtheora0 libspeex1 libopencore-amrwb0 libopencore-amrnb0 libass5 -y

Fourdee, 2015-10-30:
Fresh install, tested ok.

closed 2015-10-30 by Fourdee

thechildofroth, 2015-10-30:
Looking forward to testing v100 when it goes live - Dietpi is now surely the most comprehensive platform for audio on a Raspberry Pi.

thechildofroth, 2015-11-01:
In pursuit of minimalism I've managed to get the package list even smaller by using Java downloaded directly from Oracle rather than the Raspbian package (which drags a whole load of unnecessary X11 stuff in with it and I've not found the headless version yet).

Steps (care of: http://mj2p.co.uk/a-raspberry-pi-subsonic-jukebox-using-java-8/):

mkdir -p /opt and then unpack the file you just downloaded into that directory
sudo tar zxvf [file you just downloaded] -C /opt

You now need to tell the system that a new Java version is installed and that you want to use it as default with these commands

sudo update-alternatives --install "/usr/bin/java" "java" "/opt/jdk1.8.0/bin/java" 1
sudo update-alternatives --set java /opt/jdk1.8.0/bin/java

The system now knows where Java is installed but we will need to let subsonic know where to find it. To do this we update the JAVA_HOME environment variable which can be set in your environment file.

sudo nano /etc/environment and add the line
JAVA_HOME="/opt/jdk1.8.0"

You also need to point your bash console to the new environment variable.
nano ~/.bashrc add the lines
export JAVA_HOME="/opt/jdk1.8.0"
export PATH=$PATH:$JAVA_HOME/bin

Once subsonic is installed it needs to be directed to the correct sound device:

nano /usr/bin/subsonic. find the line that says -verbose:gc \ and add the following line above it
-Djavax.sound.sampled.SourceDataLine=#ALSA \

Fourdee, 2015-11-01:
Loving the pursuit of minimalism :+1: 

I was able to achieve your goal by using this on Ubuntu 14.04 (Odroid c1/xu4 images). I've updated the dietpi-software code for this:
> apt-get install oracle-java8-installer  --no-install-recommends
The following NEW packages will be installed:
java-common oracle-java8-installer

As for Debian, they still include x packages:
> apt-get install oracle-java8-jdk --no-install-recommends
The following NEW packages will be installed:
  fontconfig fontconfig-config fonts-dejavu-core libasound2 libasound2-data
  libatk1.0-0 libatk1.0-data libavahi-client3 libavahi-common-data
  libavahi-common3 libcairo2 libcups2 libdatrie1 libfontconfig1 libfreetype6
  libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgraphite2-3 libgtk2.0-0
  libgtk2.0-common libharfbuzz0b libjasper1 libjpeg62-turbo libpango-1.0-0
  libpango1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpangox-1.0-0
  libpangoxft-1.0-0 libpixman-1-0 libpng12-0 libthai-data libthai0 libtiff5
  libx11-6 libx11-data libxau6 libxcb-render0 libxcb-shm0 libxcb1
  libxcomposite1 libxcursor1 libxdamage1 libxdmcp6 libxext6 libxfixes3 libxft2
  libxi6 libxinerama1 libxml2 libxrandr2 libxrender1 libxtst6 oracle-java8-jdk
  shared-mime-info ucf x11-common

What you're proposing sounds great and i'am all for optimizations, however, it seems overkill. 
Considering Java is a shared library resource, i'd like to leave it alone as much as a possible. This is to prevent unnecessary bugs and increased work load for debugging various amount of software that uses Java. :)

FFmpeg was an exception, because it wasn't available in all repos across DietPi devices, we had to make it available :)