MichaIng/DietPi #88: Pi not updating to v96.

olfdag, 2015-10-13:
I run the dietpi-update command and everything works and it tells me to reboot but once I reboot it displays v94 and that there is a new update

Fourdee, 2015-10-13:
Hi,

I am unable to replicate your issue. Could you please send me a dietpi-bugreport.

Simply run: dietpi-bugreport
Then, reply with your reference CODE.

Thanks

olfdag, 2015-10-13:
b827eb30e7ae-0

Fourdee, 2015-10-13:
> b827eb30e7ae-0

It appears we have a few issues:
- The /etc/bash.bashrc file is missing all the DietPi aliases. By removing them, the commands wont work.
- I'am seeing you have modified the /etc/fstab file. Breaking the /var/log, /tmp mounts.

I may be wrong, and, you could have a bizarre SD card corruption, or, you've been a little too "trigger happy" with core system settings that DietPi relies on.
Regardless, I would suggest writing a new DietPi image.

closed 2015-10-13 by olfdag