MichaIng/DietPi #90: General | Install Fail2Ban for all DietPi systems, as default.

Fourdee, 2015-10-13:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=123&p=436#p428

NB: prevents brute-force attacks by banning IP address

Fourdee, 2015-10-13:
> The following NEW packages will be installed:
  fail2ban libpython-stdlib libpython2.7-minimal libpython2.7-stdlib
  libsqlite3-0 mime-support python python-minimal python2.7 python2.7-minimal
0 upgraded, 10 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,377 kB of archives.
After this operation, 16.6 MB of additional disk space will be used.

Pretty chunky, will add as an optional installation option in DietPi-Software.

Fourdee, 2015-10-13:
####Requires rsyslog (so SSH can log to /var/log/auth.log), and SSh server logging

apt-get install fail2ban --no-install-recommends -y

cat << _EOF_ > /etc/fail2ban/fail2ban.conf
[Definition]
# loglevel #1=error #2=warn #3=info
loglevel = 3
logtarget = /var/log/fail2ban.log
socket = /var/run/fail2ban/fail2ban.sock
pidfile = /var/run/fail2ban/fail2ban.pid
_EOF_

Fourdee, 2015-10-13:
Done, available in next release (v97).

closed 2015-10-13 by Fourdee

joaofl, 2019-04-20:
@Fourdee 
I tried to install it today and it did not work out of the box. Investigating, I got to know  `rsyslog` was required, and it is still not on the install script. Should I add it?

MichaIng, 2019-04-20:
@joaofl 
`rsyslog` is not required as long as journald is used to check for SSH authentication failures. And our installer pre-configures fail2ban to do so.

However does the installer break at some point? Will try it on test system.

MichaIng, 2019-04-20:
@joaofl 
It is really only the loglevel settings that you need to change (or remove to revert to defaults):
In `/etc/fail2ban/fail2ban.log`:
`loglevel = 3` => `loglevel = INFO`

I reworked our installed a bid and fixed this: https://github.com/MichaIng/DietPi/pull/2727

joaofl, 2019-04-20:
@MichaIng Thanks for the quick fix. I'll give it a try soon.