MichaIng/DietPi #95: bug report 0013ef1019eb-0 : Can't connect deluge deamon on windows client remote

Nibeck1309, 2015-10-16:
I can't connect to my RPI2 dietpi V97 deluge with my Windows 10 deluge client.

When I configure the client, the deamon appear not start
I can connect to the webui

Fourdee, 2015-10-17:
Hi,

Thanks for the bug report.
I'am unable to replicate your issue.
[image]

Have you tried restarting the service as per the 1st run guide? http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&start=10#p61

dietpi-services restart

Nibeck1309, 2015-10-20:
Hy,

I do a clen install of dietpi, install deluge from dietpi-software, reboot, dietpi-service restart. 
I can connect to deluge from windows deluge client.
On next reboot i can't connect. The deluge deamon appear not run on windows deluge

Fourdee, 2015-10-20:
I will run another test today on my RPi v1 and Wheezy (which i believe, from your bug report is your system setup) using your steps to try and replicate the issue.

I'll be in touch.

Nibeck1309, 2015-10-20:
Hy,

I have news information,

When I reboot my RPI, on next reboot I can't connect to deluge with windows deluge client.
But if after I reboot Dietpi, i executed dietpi-service restart, I can connect to deluge....

Why I do execut dietpi-service restart for deluge works after i reboot Dietpi ?

Fourdee, 2015-10-20:
Thanks for the update. 
I'am just waiting for my RPi v1 to install deluge (busy day), then i'll take a look.

Fourdee, 2015-10-20:
Using client: 1.3.12-1 & Python 2.7
http://download.deluge-torrent.org/windows/py2.7/deluge-1.3.12-1-win32-py2.7-setup.exe

After reboot, unable to connect deluge client:
- Only occurs on Wheezy. Jessie is fine.
- Log file is identical, after reboot and restart of services.
[deluge_after.txt](https://github.com/Fourdee/DietPi/files/17055/deluge_after.txt)
[deluge_reboot.txt](https://github.com/Fourdee/DietPi/files/17056/deluge_reboot.txt)

Fourdee, 2015-10-20:
I will close this ticket. Please see my previous post for resolution options.

Please re-open ticket if required.

closed 2015-10-20 by Fourdee