MichaIng/DietPi #97: MT7601 wifi dongle

Blamoo, 2015-10-23:
Is tehre something wrong with the drivers for this device?

I connect it to my Pi, fresh DietPi v94 Jessie install, no optimized software.

dmesg gives me this:
```
[  323.845154] usb 1-1.4: new high-speed USB device number 8 using dwc_otg
[  323.956490] usb 1-1.4: New USB device found, idVendor=148f, idProduct=7601
[  323.956514] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  323.956531] usb 1-1.4: Product: 802.11 n WLAN
[  323.956546] usb 1-1.4: Manufacturer: MediaTek
[  323.956562] usb 1-1.4: SerialNumber: 1.0
[  324.035135] usb 1-1.4: reset high-speed USB device number 8 using dwc_otg
[  324.138207] mt7601u 1-1.4:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  324.139714] mt7601u 1-1.4:1.0: Direct firmware load for mt7601u.bin failed with error -2
[  324.140411] mt7601u: probe of 1-1.4:1.0 failed with error -2
```

I copied that file `mt7601u.bin` from the official Raspbian to `/lib/firmware` and then it worked:
```
[   32.405209] usb 1-1.4: new high-speed USB device number 5 using dwc_otg
[   32.516566] usb 1-1.4: New USB device found, idVendor=148f, idProduct=7601
[   32.516589] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   32.516606] usb 1-1.4: Product: 802.11 n WLAN
[   32.516622] usb 1-1.4: Manufacturer: MediaTek
[   32.516638] usb 1-1.4: SerialNumber: 1.0
[   32.612980] cfg80211: Calling CRDA to update world regulatory domain
[   32.795214] usb 1-1.4: reset high-speed USB device number 5 using dwc_otg
[   32.898218] mt7601u 1-1.4:1.0: ASIC revision: 76010001 MAC revision: 76010500
[   32.906620] mt7601u 1-1.4:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[   33.306414] mt7601u 1-1.4:1.0: Warning: unsupported EEPROM version 0d
[   33.306438] mt7601u 1-1.4:1.0: EEPROM ver:0d fae:00
[   33.558164] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   33.559898] usbcore: registered new interface driver mt7601u
```

Is this supposed to work like this?

septs, 2015-10-23:
the device is "小米随身WiFi" or "360随身WiFi" or other?

Fourdee, 2015-10-23:
What Kernel/Firmware version and distro, was Raspbian running when you copied  /lib/firmware/mt7601u.bin?

Have you tried updating the RPi Kernel/Firmware (rpi-update) on your DietPi system?

Blamoo, 2015-10-23:
> What Kernel/Firmware version and distro, was Raspbian running when you copied /lib/firmware/mt7601u.bin?

I'm not sure, I can't check it right now...

> Have you tried updating the RPi Kernel/Firmware (rpi-update) on your DietPi system?

yes, nothing happened...

Also, that `mt7601u.bin` can be found here: https://github.com/porjo/mt7601/blob/master/src/mcu/bin/MT7601.bin

Fourdee, 2015-10-24:
Thanks.
I've created a patch for DietPi Jessie:https://github.com/Fourdee/DietPi/blob/testing/dietpi/patch_file#L816-L820

This will be available and applied in the DietPi v99 update.

Blamoo, 2015-10-24:
Thanks, i'll test it ASAP :)

Fourdee, 2015-10-27:
I'am due to release v99 later this week. Once you've updated, you can test it ;)

closed 2015-10-27 by Fourdee

Blamoo, 2015-10-28:
It works!