MichaIng/DietPi #98: DietPi-Config | Add i2c/spi and other hardware options for RPi's

Fourdee, 2015-10-24:
> http://fuzon.co.uk/phpbb/posting.php?mode=quote&f=9&p=560#pr560

Fourdee, 2015-10-26:
Notes:
Patch, add to config.txt
> if (( $(cat /DietPi/config.txt | grep -ci -m1 'dtparam=i2c_arm') == 0 )); then
	echo -e "\n#i2c settings" >> /DietPi/config.txt
	echo -e "#dtparam=i2c_arm=on" >> /DietPi/config.txt
fi
if (( $(cat /DietPi/config.txt | grep -ci -m1 'i2c_arm_baudrate=') == 0 )); then
	echo -e "#i2c_arm_baudrate=400000" >> /DietPi/config.txt
fi

i2c ON
>apt-get install python-smbus i2c-tools -y
if (( $(cat /etc/modules | grep -ci -m1 'i2c-bcm2708') == 0 )); then
	echo -e "i2c-bcm2708" >> /etc/modules
fi

i2c OFF
>sed -i '/i2c-bcm2708/d' /etc/modules &> /dev/null

Fourdee, 2015-10-27:
Done. Available in v99.
https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-rpi_i2c_control
[image]

erobinson-pb, 2019-10-26:
I'm seeing i2c, not SPI

incanus, 2021-02-22:
Same here, latest (`v6.34.3`) only shows I2C.

MichaIng, 2021-02-22:
```
G_CONFIG_INJECT 'dtparam=spi=' 'dtparam=spi=on' /boot/config.txt
```

incanus, 2021-02-22:
Sure, I know how to do it. My point is that, as stated above, the config tool which this issue was opened about does not in fact allow for SPI configuration as stated in the fix.

MichaIng, 2021-02-23:
I just added the option: https://github.com/MichaIng/DietPi/pull/4138

Support for other devices can be easily added when they ship with device tree overlays.

Also there are quite some options and alternative SPI interfaces available on RPi. E.g. RPi 4 theoretically has 7 different SPI interfaces which can use 1-3 chip select lanes. I don't understand these very well currently, but the official RPi documentation is quite clear, so we could add further options to adjust those, based on RPi model. However, the default SPI0 with 2 cs lanes is by far the most used one, which can be seen by some LCD installers which add `dtparam=spi=on` only.

But that's for a later release.

closed 2021-02-28 by MichaIng