MichaIng/DietPi #99: Driver for DWA-182 rev. C

lukecyca, 2015-10-25:
This wifi module has linux drivers [from the manufacturer](http://support.dlink.com/ProductInfo.aspx?m=DWA-182). I attempted to download and compile them, but it's not clear to me where to get the correct `linux-headers` from. There doesn't seem to be a package available.

I am running DietPi V98, and have done `rpi-update` to get the latest firmware.
```
root@benchpi:~# uname -a
Linux benchpi 4.1.11+ #822 PREEMPT Fri Oct 23 16:14:56 BST 2015 armv6l GNU/Linux
```

My first priority would be to figure out how to compile these drivers. Afterwards it would be nice to have them included in DietPi for others' benefit.

Fourdee, 2015-10-26:
Hi Luke,

Yep, from the download like you provided, looks like i'll need to compile the driver. I'll attempt this in the next few days and keep you updated.

Does the official Raspbian support this device?

lukecyca, 2015-10-26:
Thanks for the reply, Dan.

I have not tried with Raspbian. I suspect the drivers are not built-in, but it would be fairly easy to install `linux-headers-rpi` and then build the drivers as per the manufacturer's instructions.

lukecyca, 2015-11-02:
Anything I can do to help you with this?

If I could get the kernel headers installed, then I could try compiling the vendor's drivers to ensure they work.

Fourdee, 2015-11-02:
Hi Luke,

Apologies for the delay, extremely busy at the moment with v100. 
Yes, if you could try compiling and testing, seems the most logical as I dont own DWA-182.

You can search Apt for a list of packages:
apt-cache search linux-headers
As for which one you need, not sure, but i'd assume linux-headers-rpi.

lukecyca, 2015-11-04:
I managed to get this build for my `4.1.10+` kernel (DietPi Jessie, updated). It wasn't straightforward so I'll post my instructions here.

1. `apt-get install gcc-4.8 g++-4.8`
1. Used [rpi-source](https://github.com/notro/rpi-source) to download the correct kernel sources and set things up for compiling a kernel module. This utility is awesome! It automates the frustrating process of getting the correct headers since the typical `linux-headers-*` packages don't exist for most rasbian kernels.
1. I had no luck with the [manufacturer's driver source](http://support.dlink.com/ProductInfo.aspx?m=DWA-182). Instead I used [rtl8812AU_8821AU_linux](https://github.com/abperiasamy/rtl8812AU_8821AU_linux).
1. After building and installing as per the README in that driver, `modprobe cfg80211 8812au` worked and the wireless device shows up in `ifconfig -a`.

I'm set for now, but it would be great to include this in future versions of DietPi!

Fourdee, 2015-11-04:
Hi Luke,

Thanks for the info, and great to hear the issue is resolved.

If the DWA-182 works on the official raspbian, i'll add support for it in DietPi also. If it does not, you will need to manually compile the driver as required.
The amount of time required to support all DietPi images with a user specific/unsupported driver would be a nightmare.

closed 2015-11-04 by Fourdee