MichaIng/DietPi #102: RPi Jessie | WiFi = excessive "calling CRDA" print on tty1

Fourdee, 2015-10-27:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=151#p565

Fourdee, 2015-10-29:
> bugreport-c4e98412b213-0

wlan0 adapter is not being brought up during boot.

editing /etc/rc.local and adding the following is a workaround:
ifup wlan0

temp fix. Possible systemd ordering.

Fourdee, 2015-10-30:
As per http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=151#p568

Regardless of the /etc/default/crda setting. CRDA claims its not set:
root@dietpi:~# crda
COUNTRY environment variable not set

Notes to self:
- Can we disable CRDA?
- Can we run CRDA using runtime options (eg: crda --code=GB)?
- Do we need to add anything to the /etc/network/interfaces file for CRDA to function?

Blamoo, 2015-11-03:
`apt-get install iw crda` solves this, I think...

Fourdee, 2015-11-03:
Spot on, resolves the issue! Thanks for the info.

iw and crda will be installed during v100 patch.

Date: Tue, 3 Nov 2015 02:24:17 -0800

Blamoo, 2015-11-03:
Ah, sorry for deleting that post :p

closed 2015-11-08 by Fourdee