MichaIng/DietPi #106: dietpi-software fails when dpkg is on interrupted state

Blamoo, 2015-11-03:
`dietpi-software` marks software as installed, but won't install anything unless you run `dpkg --configure -a`
Even with this error, it still says the software is installed on `dietpi-software` and `dietpi-uninstall`.

Fourdee, 2015-11-04:
Hi Blamoo,.

Sounds like a previous apt-get install failed, this would cause dpkg to prevent further installations, until the issue is resolved.

I'll need to check your settings, can you send me a bug report please:
dietpi-bugreport

Also, can you remember which software you installed, prior to the error being displayed?

Notes to self:
- Check dpkg for errors before dietpi-software runs installations.
- Check /DietPi/dietpi/.installed against dpkg -l  to confirm installed.

Blamoo, 2015-11-04:
k, I can't do it right now but I'll try to reproduce it tonight.

Fourdee, 2015-11-05:
I've added some error checking to v100. This will inform the user and provide the apt error code when this occurs.

closed 2015-11-08 by Fourdee