MichaIng/DietPi #109: Upgrade "v100" been waiting for "apt-get update"

septs, 2015-11-09:
[image]

Model: Raspberry Pi 1 model B+

Fourdee, 2015-11-09:
Hi Septs,

At the command prompt, can you run the following code please:
> killall apt-get
> /DietPi/dietpi/dietpi-apt-get_update 2

septs, 2015-11-09:
thank you, `rm /DietPi/dietpi/.dietpi-apt-get_update` after normal

Fourdee, 2015-11-09:
Awesome, thanks for letting me know.

I've updated the dietpi-apt-get_update code for v101. This will prevent this issue from occurring.

closed 2015-11-14 by Fourdee

septs, 2016-07-08:
[image]

Fourdee, 2016-07-08:
@septs 

Can you please paste results of:
```
cat  /DietPi/dietpi/.dietpi-apt-get_update
```

And:
```
/DietPi/dietpi/dietpi-apt-get_update 2 
#Your connection looks slow, please wait for at least 2 minutes before terminating.
cat /var/log/dietpi-apt-get_update
```

septs, 2016-07-08:
later, several attempts by the update

-----------------------------------------

Re: record screen: 
Ubuntu see https://wiki.ubuntu.com/CreatingScreencasts
Windows/OS X see http://www.cockos.com/licecap/

Fourdee, 2016-07-08:
Nice:
[image]