MichaIng/DietPi #111: kodi don't start

Nibeck1309, 2015-11-11:
Hello,

Reference CODE: b827eb45e355-0

I'm using dietpi-sofware for installing kodi.

When i want to launch it, i have this error message :+1: 

/DietPi/dietpi/misc/rpi_kodi_noblackscreen_run: ligne 25: kodi-standalone : commande introuvable

Thanks for your help

Fourdee, 2015-11-11:
Hi Nibeck1309,

Thanks for the report.

It appears you are using additional/modified apt sources that are causing your issue. 
Your system is installing kodi: 5:14.2+repack-dmo1b1 , instead of 2:14.2-1wheezy from the official Raspbian Jessie Repo.

Please check your custom apt sources, disabling any that are overriding the Kodi package from the Raspbian repo. Then reinstall Kodi.

Fourdee, 2015-11-11:
Update: 
It appears Kodi is being installed from the debian-multimedia repo (which is installed during FFmpeg). I'll look into this further.

Nibeck1309, 2015-11-11:
OK thanks for report.

Fourdee, 2015-11-12:
Ok this will be fixed for v101. After installing FFMPEG, the deb.multimedia repo will be removed.

You will need to manually re-install Kodi on your system, please do the following:
- use dietpi-uninstall and select Kodi for removal
- run below in terminal to remove deb.multimedia repo
> sed -i '/www.deb-multimedia.org jessie main non-free/d' /etc/apt/sources.list
/DietPi/dietpi/dietpi-apt-get_update 2

- Now you can run dietpi-software and select Kodi for installation.

Nibeck1309, 2015-11-12:
Hello,

Thanks for answer.

use dietpi-uninstall and select Kodi for removal
run below in terminal to remove deb.multimedia repo

sed -i '/www.deb-multimedia.org jessie main non-free/d' /etc/apt/sources.list
/DietPi/dietpi/dietpi-apt-get_update 2
Now you can run dietpi-software and select Kodi for installation

That works perfectly.

I'm install kodi and it work

Thanks

closed 2015-11-14 by Fourdee