MichaIng/DietPi #115: bug : dietpi orange pi pc module load

keiser1080, 2015-11-15:
Hi,

there are some problem with the module loading
```
root@DietPi:~# lsmod
Module                  Size  Used by
root@DietPi:~# cat /etc/modules
#Orange Pi -------------
#For SATA Support
sw_ahci_platform

#Display and GPU
lcd
hdmi
ump
disp
mali
mali_drm
8192cu
8188eu
8189es
gpio-sunxi

#bt
btusb
root@DietPi:~# modprobe btusb
modprobe: ERROR: could not insert 'btusb': Exec format error
root@DietPi:~# modprobe -f btusb
```

Fourdee, 2015-11-15:
Yep its a known issue. 
For the Orange Pi we are using a modified version of http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=342 image as the base. The original image also has the same issues.

keiser1080, 2015-11-16:
did you use the same kernel?
i  try to found some hep http://www.orangepi.org/orangepibbsen/forum.php?mod=redirect&goto=findpost&ptid=342&pid=6496&fromuid=135990

keiser1080, 2015-11-18:
the kernel is hosted somewhere else see the attached script.
[update_kernel.sh.txt](https://github.com/Fourdee/DietPi/files/38185/update_kernel.sh.txt)

Fourdee, 2015-11-18:
Just ran the script. Modules are now working :+1: 

> root@DietPi:~# lsmod
Module                  Size  Used by
gpio_sunxi              8233  0
8189es                901572  0
8188eu                915632  0

You can do the following:
```
wget https://raw.githubusercontent.com/loboris/OrangePi-BuildLinux/master/update_kernel.sh
chmod +x update_kernel.sh
./update_kernel.sh
```

Fourdee, 2015-11-18:
I've added the update script to dietpi-config > Advanced > update firmware.

keiser1080, 2015-11-18:
Thanks!

I have to run the script? Or you will push the update?

keiser1080, 2015-11-22:
feedback:
script it's working but return small error
cp: impossible d'évaluer « /boot/uImage »: Aucun fichier ou dossier de ce type
cp: impossible d'évaluer « /boot/script.bin »: Aucun fichier ou dossier de ce type

don't found /boot/uImage &  /boot/script.bin

But everything is working 

thanks again

closed 2015-11-30 by Fourdee