MichaIng/DietPi #116: DietPi-Software | Add Plex media server option

Fourdee, 2015-11-17:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=12&t=201&p=783#p779

Fourdee, 2015-12-15:
Notes to self:
```
wget http://dev2day.de/plex-latest -O package.deb
dpkg -i package.deb
rm package.deb
```

Service wont start:
```
root@DietPi:~# /usr/sbin/start_pms & >/dev/null 2>&1
[1] 939
root@DietPi:~# 6 3000 /root/Library/Application Support
8192
/usr/sbin/start_pms: 29: /usr/sbin/start_pms: ./Plex Media Server: not found
```
- References to en_US required.

lwittwer, 2016-05-12:
The reason it does not start is that Plex requires en_US.UTF-8 to be generated on the system which DietPi does not by default.

Fourdee, 2016-05-13:
@lwittwer 
Yep, en_GB.UTF-8 is default on DietPi.
I wonder if we can just generate the en_US.UTF-8 locale, or does it require set to default on system.

xenfomation, 2016-05-13:
A default is required, but you can switch/regenerate on the fly.  I am heading into work now, but will catch up with these items around my lunch time!

lwittwer, 2016-05-13:
Just requires to be generated. Just tested it out on my system

https://github.com/Fourdee/DietPi/issues/116#issuecomment-219011198

Fourdee, 2016-05-13:
@lwittwer @xenfomation 
Thanks for the info :+1: 

I'll mark this for v119, seems popular.

Fourdee, 2016-05-25:
#### Notes:
Original source for package doesnt have arm64 available (https://dev2day.de/pms/ ):

```
W: Failed to fetch https://dev2day.de/pms/dists/jessie/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

```echo -e "deb [arch=armhf] https://dev2day.de/pms/ jessie main" > /etc/apt/sources.list.d/plex.list```:
```
root@DietPi:~# apt-get install plexmediaserver -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 plexmediaserver : Depends: plexmediaserver-installer but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Fourdee, 2016-05-25:
Well, compiling is out the window. Looks like Plex server source isnt available on GitHub. So i'd assume Plex media server is not opensource. 

Nor do they release or provide ARM binaries (only x86_32/64): https://plex.tv/downloads

So unless: 
- anyone has another method of installing Plex media server on ARM's
- Has the ARM debian packages for Plex media server
- has the source code for the server so I can compile it
Plex media server will only available for the DietPi VM's.

Fourdee, 2016-05-27:
Done v119. 

Available for DietPi VM (Virtual machine) installations only.

Doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1949#p1949

I'll mark this as closed. However, if myself or anyone finds a working alternative for Plex Media Server on ARM platforms, please reopen and let me know so I can implement it.

Fourdee, 2016-05-27:
Turns out, although ```[armhf]``` lets us try to install ```plexmediaserver```, on ARM64, that also causes the missing ```plexmediaserver-installer``` error.
Rpi installed fine.

I'll get this added to all 32bit ARM SBC's (exlcuding Pine and C2)

Fourdee, 2016-05-27:
en_US.UTF8 installed, en_GB.UTF8 is default:
```
root@DietPi:~# /usr/lib/plexmediaserver/start.sh
/usr/lib/plexmediaserver/start.sh: 11: /usr/lib/plexmediaserver/start.sh: ./Plex Media Server: not found

root@DietPi:~# /usr/lib/plexmediaserver/Plex\ Media\ Server
/usr/lib/plexmediaserver/Plex Media Server: error while loading shared libraries: libboost_system.so.1.59.0: cannot open shared object file: No such file or directory

```

en_US.UTF8 must be the default locale on system, not just installed. For non-VM installations.

Fourdee, 2016-05-27:
Done, available for all DietPi devices, excluding ARM64 (Pine/C2).

closed 2016-05-27 by Fourdee

LukasBombach, 2016-07-31:
Hi, Pine64 User here. Any hope ARM64 will be supported at some point?

LukasBombach, 2016-07-31:
Btw, for people like me still looking to install Plex Media Server, it can be done the "normal" way still, like this: https://gist.github.com/LukasBombach/2943fc2a1cb5fd4bbd1c3e6b477ce033

Fourdee, 2016-08-02:
@LukasBombach :+1: 

> https://gist.github.com/LukasBombach/2943fc2a1cb5fd4bbd1c3e6b477ce033

Does the armhf binary work on your Pine A64, when running arm64/aarch64 OS ```uname -a```? 
If so, this indicates the kernel and ARM supports 32 bit execution. Then we can get PMS added for the Pine boards.

LukasBombach, 2016-08-03:
I'm not so Linux-savvy (I'm a programmer though, so you can tell me in technical terms), how would I go and try that?

Fourdee, 2016-08-04:
@LukasBombach 

> how would I go and try that?

Sorry, basically, i'am asking if doing this installation results in a working Plex installation? I've not had a chance to try it myself yet : 
https://gist.github.com/LukasBombach/2943fc2a1cb5fd4bbd1c3e6b477ce033

LukasBombach, 2016-08-04:
It worked for me on my Pine64 using

* The latest Ubuntu created by the Pine team
* The latest Debian created by the Pine team
* And of course DietPi

Fourdee, 2016-08-04:
@LukasBombach Done, PMS will be available for Pine in v127. Many thanks for the installation code :+1:

LukasBombach, 2016-08-04:
Woop woop, awesome!

Ardakilic, 2016-08-20:
Just a heads-up. I've installed from that dev2day repo above just fine on ODROID-XU4 to DietPi. I wonder how it's handled but the current one on DietPi is a bit behind I guess.

Fourdee, 2016-08-21:
@Ardakilic 
DietPi currently uses the dev2day repo for PMS on ARM devices: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L3978-L4039

Ardakilic, 2016-08-21:
@Fourdee Oh my mistake then, totally missed that x86_64 sign :+1: