MichaIng/DietPi #117: Error/bug on Dietpi-software

jcastro, 2015-11-20:
I'm not sure why but now the command dietpi-sofware shows up as the follows http://d.pr/i/1itEj/2i5obcwK and I'm unable to install or uninstall new software from here. Please let me know how I can help you debug the error, I'm not sure what I could send you to help!

Thanks for this great piece of software, loving it!

jcastro, 2015-11-20:
http://d.pr/i/1iJAV/3YpKhOo1

Fourdee, 2015-11-20:
Hi,

I'd like to get more info on your installation, please can you run :
dietpi-bugreport 
and paste the reference CODE. No need to fill out the text.

The1st looks like a character / UTF8 issue. Did you change locale on the system?

The 2nd looks like apt cache is empty or corrupt. Please try running the following before dietpi-software:
/DietPi/dietpi/dietpi-apt-get_update 2

jcastro, 2015-11-20:
of course! code is b827eb49e69b-0 

I'm not sure if I changed it, but it's also not working completely. If I enable an app I get the error as in http://d.pr/i/1iJAV/3YpKhOo1 and if I disable apps it won't do anything when going to "Start install". It's like it's not remembering the option. 

I just tried your command /DietPi/dietpi/dietpi-apt-get_update 2 but it doesn't seem to fix the problem!

Thanks a lot!

Fourdee, 2015-11-20:
Thanks,

Your DietPi settings are fine. I also tested the RPi GPIO installation on a fresh Wheezy image, no issues.

Ok, I'll need some more info on your system:
- How are you using Pi-hole on your network? (eg: dns setting on router, or, manual dns entry per device.)

Please run the following:
```
mkdir -p /root/send
cat /etc/locale.gen > /root/send/locale
apt-get install -q mc -y &> /root/send/apt-get_install
apt-get update -q &> /root/send/apt-get_update
apt-get purge mc mc-data -y
```
Can you then upload the files under /root/send . Zip or as text attachments, doesnt matter.

jcastro, 2015-11-20:
I was about to copy the send.zip and I somehow broke the inittab... now I can't connect to the Pi.. I will need to reinstall :(. Sorry I can't send the files

About DNS, I have them setup in the router configuration

Fourdee, 2015-11-20:
No worries. Let me know if the issue continues after applying the below.

If using Pi-hole as the DNS server address on your router, you must enable a different DNS address for the Pi-hole device.

Add the following to /etc/network/interfaces, under your iface device:
```
# Google DNS servers
dns-nameservers 8.8.8.8 8.8.4.4
```

eg:
```
#/etc/network/interfaces

# Local
auto lo
iface lo inet loopback

# Ethernet
#auto eth0
iface eth0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
# Uncomment for Google DNS servers
dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
#auto wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-essid FuzonWifi
wireless-key abcde12345
wireless-mode Managed
wireless-power off
wpa-ssid FuzonWifi
wpa-psk abcde12345
```

Fourdee, 2015-11-20:
Also, as your using Wheezy. Run the following to update networking:
```
/etc/init.d/networking reload
/etc/init.d/networking restart
```

Fourdee, 2015-11-23:
Please reopen if required.

closed 2015-11-23 by Fourdee