MichaIng/DietPi #119: DietPi-Software | DietPi-Cam not starting during boot.

Fourdee, 2015-11-23:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=11&t=218
Reference Code: 74da383582db-0
su -c 'raspimjpeg > /dev/null &' www-data
This account is currently not available.

> RPi 2 - Jessie

Fourdee, 2015-11-23:
Resolved for V102, using sudo and defining user.

sudo -u www-data raspimjpeg > /dev/null &
sudo -u www-data php /var/www/dietpicam/schedule.php > /dev/null &

R1CH-T, 2015-11-23:
Thanks Dan. :+1: 
I did an "apt install sudo" then changed the offending lines in "/etc/raspimjpeg_init" as above, ran "/etc/raspimjpeg_init start" and it worked like a charm.
I rebooted and the camera was broadcasting right away.

How come the script chokes up on "su -c", though, and not sudo?
Strikes me as a little odd.

Fourdee, 2015-11-23:
I have absolutely no idea.
Its one of those "it was working yesterday" things. I can't see anything I've changed that would cause this issue.

Maybe a repo package got updated, wouldnt be the first time apt-get upgrade broke something.

closed 2015-11-30 by Fourdee