MichaIng/DietPi #120: DietPi-Software | DietPi-Cam - Motion detection not recording.

Fourdee, 2015-11-23:
>http://fuzon.co.uk/phpbb/posting.php?mode=quote&f=11&p=854#pr854

> Motion detection starts up, it shows up in the log file and the motion service is listed as running on the system, but the weird thing is that motion.php page is utterly blank and no recording is being triggered.
I'm really just stumbling around in the dark here. I've been working my way through http://elinux.org/RPi-Cam-Web-Interface but it's quite dense reading and doesn't really have a quickstart section, explaining the basics.
i'd have a proper go at reading the forum thread at https://www.raspberrypi.org/forums/view ... 43&t=63276 but it's 121 pages long!

Fourdee, 2015-11-28:
Jessie: motion.php blank.
Wheezy: ..................

Note to self:
 - Update DietPi-Cam to latest RPi-Camweb

Fourdee, 2015-11-28:
Confirmed working on Wheezy. Appears to be a bug in RPi-cam-web-interface on Jessie.

Will attempt to update DietPi-Cam version to latest RPi-cam-web-interface version for a resolution.

Fourdee, 2015-12-04:
I've updated DietPiCam to match the current RPi cam control (v6.0.19), but, still having a blank page on jessie for motion.

Turns out this is a known issue for Jessie: https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/214#issuecomment-152719047
Looks like the cause is with the motion package installation. No doubt the libraries are different to Wheezy.

R1CH-T, 2016-05-11:
Hey Dan,

Lsst week, I upgraded to v117 and had another go at setting up DietPiCam (on Jessie), but I'm still experiencing the "blank motion detection page" problem.

I am also still having some problems getting motion detection working at all.

I notice that there have been some bug reports and minor tweaks to this module recently and thought that maybe, if you get the chance this week; could you pull in the latest version and add it to a testing branch, so that I can test it this weekend and maybe have a go at trying to get motion detection working for me?

Sorry to bother you, but I'd like to see this working properly on the mainline DietPi.
If you don't have the time, I completely understand.

Thanks,
Rich.

Fourdee, 2016-05-11:
@R1CH-T 

Yep no worries. I need to update the installation anyway, hopefully the motion detection has been fixed by the RPiwebcam devs.

I'll mark this for v119 (https://github.com/Fourdee/DietPi/milestones/v119). Will take a look once v118 (https://github.com/Fourdee/DietPi/milestones/v118) is completed and released.

Fourdee, 2016-05-18:
Updated to 6.1.6 

Internal motion detection working well:
[image]

External motion = error when loading motion options:
```
(mod_fastcgi.c.2695) FastCGI-stderr: PHP message: PHP Warning:  file_get_contents(http://127.0.0.1:6642/0/config/list): failed to open stream: Connection refused in /var/www/dietpicam/motion.php on line 25
```

Fourdee, 2016-05-18:
I've also managed to remove 2 conf files from DietPi-Ramdisk. Should free up some memory.
https://github.com/Fourdee/DietPi/commit/a1e5d704e25133f81c2ea85b3775439d8a936e5b

closed 2016-05-20 by Fourdee

R1CH-T, 2016-05-23:
Hi.
I just changed to the "Testing" branch, updated to v119 and installed DietPiCam.
I can confirm that internal motion detection now works flawlessly. Fantastic work!
I may need to play around with it and adjust the sensitivity or maybe create a mask, but apart from that, it's working great.
Thanks for your efforts, Dan.
:+1: