MichaIng/DietPi #121: Documentation request: Kernel config (docu), if applicable

cprior, 2015-11-25:
DietPi should document the kernel configuration as it is hard to troubleshoot:

- is it the upstream kernel?
- or are there DietPi-specific changes

If the latter, access to /proc/config.tgz is probably the easiest way.
(Yes, there is also the script extract-ikconfig in the kernel sources, but that is probably now widely known.)

Background:
I am currently investigating why on Jessie the serial console doesn't work despite systemd an /boot/cmdline.txt modifications.
Without several SD cards, a cross compile environment and/or several boards I am currently down to recompiling the kernel with serial console support, just to see if that makes a difference.

But I can think of other situations where definitive knowledge about the kernel config might come in handy.

Fourdee, 2015-11-25:
Hi,

DietPi uses the standard Kernel that came from the image. We do nothing to them.

cprior, 2015-11-25:
Thanks for the clarification regarding the kernel. Upstream I found Issue https://github.com/Hexxeh/rpi-update/issues/185 which said
>It is built as a module now.
>`sudo modprobe configs`
>should make it appear.

Regarding the serial console I also get the /sbin/agetty process as per Systemd unitfile with journalctl.
I need to troubleshoot that later (although the cable works with a MinnowBoard Max I ordered a replacement for comparisons and I will also test with a stock Raspbian).

The question about the kernel config is solved though -- many thanks!

closed 2015-11-25 by cprior