MichaIng/DietPi #122: problems with ownership of /var/log/mongodb/mongodb.log

otmezger, 2015-11-26:
Hi, 

I'm using DietPi-Ramlog # 2 config.

I installed mongo using 
    
    sudo apt-get install mongoldb

every time the system boots, I see this log from mongoldb:

```
energy@DietPi:/etc/init.d$ sudo service mongodb status
● mongodb.service - An object/document-oriented database
   Loaded: loaded (/lib/systemd/system/mongodb.service; enabled)
   Active: failed (Result: exit-code) since Thu 2015-11-26 17:19:00 CST; 7min ago
     Docs: man:mongod(1)
  Process: 367 ExecStart=/usr/bin/mongod --config /etc/mongodb.conf (code=exited, status=1/FAILURE)
 Main PID: 367 (code=exited, status=1/FAILURE)

Nov 26 17:18:58 DietPi systemd[1]: Started An object/document-oriented database.
Nov 26 17:19:00 DietPi mongod[367]: all output going to: /var/log/mongodb/mongodb.log
Nov 26 17:19:00 DietPi mongod[367]: can't open [/var/log/mongodb/mongodb.log] for log file: errno:2 No such file or directory
Nov 26 17:19:00 DietPi mongod[367]: Bad logpath value: "/var/log/mongodb/mongodb.log"; terminating.
Nov 26 17:19:00 DietPi systemd[1]: mongodb.service: main process exited, code=exited, status=1/FAILURE
Nov 26 17:19:00 DietPi systemd[1]: Unit mongodb.service entered failed state.
energy@DietPi:/etc/init.d$ 
```

If I check the permissions of `/var/log/mongodb/mongodb.log` I can see they are correct:

```
energy@DietPi:/etc/init.d$ ls -lah /var/log/mongodb/
total 0
drwxrwxr-x 2 mongodb mongodb  60 Nov 26 17:19 .
drwxrwxr-x 8 root    root    340 Nov 26 17:19 ..
-rwxrwxr-x 1 mongodb mongodb   0 Nov 26 17:19 mongodb.log
energy@DietPi:/etc/init.d$ ls -lah /var/log/mongodb/mongodb.log 
-rwxrwxr-x 1 mongodb mongodb 0 Nov 26 17:19 /var/log/mongodb/mongodb.log
```

Before the ownership where mongoldb, I needed to add the line 
    chown -R mongodb:mongodb /var/log/mongodb &> /dev/null

under the function `Apply_Unique_Logfile_Permissions` in the file `/DietPi/dietpi/dietpi-ramlog`

I also changed the ownership  of the copy of the ram disk in boot folder with 

    chown -R mongodb:mongoldb /root/logfile_storage/mongodb

Now the problem is, as you can see, that after boot, mongodb didn't start. The strange thing is that if I run `sudo service mongodb start` it will start straight away without any problems. So basically I'm having some ownership problems during boot, where mongodb is maybe starting before the whole ram log logic ends. 

How can I fix this? 

thanks

blindpet, 2015-11-27:
If you are on jessie you can adjust the systemd script for mongodb to start after ramlog. Something like this sudo code

After=ramlog.service

Fourdee, 2015-11-27:
Blindpet is right. Looks like Mongodb is running before dietpi-ramlog has generated /var/log.

Add the following to the systemD service of mongodb:
```
After=dietpi-service.service
```

otmezger, 2015-11-27:
Hi there, 

thanks for the reply, where do I find the systemD service of mongodb?

otmezger, 2015-11-27:
is this the file: `/lib/systemd/system/mongodb.service` and if yes, on which category should I put the line you told me?

my file looks like this:
```
[Unit]
Description=An object/document-oriented database
Documentation=man:mongod(1)

[Service]
User=mongodb
ExecStart=/usr/bin/mongod --config /etc/mongodb.conf

[Install]
WantedBy=multi-user.target
```

otmezger, 2015-11-27:
Ok, I moved the line around and if I put it as last line in section `[Unit]` then it'll work. looks like magic to me, and I don't know if this is wat needed to be done, but it seems to be working now.

closed 2015-11-27 by otmezger

Fourdee, 2015-11-27:
```
cat << _EOF_ > /lib/systemd/system/mongodb.service
[Unit]
Description=An object/document-oriented database
Documentation=man:mongod(1)
After=dietpi-service.service

[Service]
User=mongodb
ExecStart=/usr/bin/mongod --config /etc/mongodb.conf

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
```

Fourdee, 2015-11-27:
>Ok, I moved the line around and if I put it as last line in section [Unit] then it'll work. looks like magic to me, and I don't know if this is wat needed to be done, but it seems to be working now.

Yep, you beat me to it :)

otmezger, 2015-11-27:
excellent help @Fourdee, Now i'm confident that it won't cause any trouble somewhere else. Thanks for doing Dietpi, for you and the community.

Fourdee, 2015-11-27:
No worries, if you experience any further issues, just give us a shout and/or reopen ticket :)