MichaIng/DietPi #125: General | Wget SSL error for some users

Fourdee, 2015-11-28:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=11&t=214#p806

Resolution:
Convert wget to curl throughout all DietPi-Scripts.

blindpet, 2015-11-28:
Just to let you know this is definitely a network error and not your image. A user had the same error on Minibian grabbing a file from github. I could run the same exact command and grab the file without the error using the same image on my Pi at home.

Fourdee, 2015-11-28:
Thanks for the info Blindpet :+1: 

I thought as much. Still, converting to curl isnt a bad idea, it also seems to follow redirects unlike wget.

Fourdee, 2015-11-29:
Replacing:
wget --spider

The only way to "test" a connection with curl is:
```
curl --head
```
However, the exit code is always 0, regardless of HTML status code (eg: 404).

We can use a grep check:
```
curl --head www.test.com/file.zip | grep -ci -m1 200
```
However, any other status code would return a failure (eg: redirect). Also, checking for 200 is risky due the vast output from curl.

I'am unable to find a "viable" replacement for wget --spider at the moment. If only curl had an option to "not download" the content/file, if only..

Fourdee, 2015-11-30:
```curl --trace /dev/null https://raw.github.com &> /dev/null;echo $?```
Returns usable exit codes.

Fourdee, 2015-12-01:
Ok, for v102:
DietPi-Update has changed from wget to curl. At the very least, this will allow users with wget failures to install/update DietPi.

As the issue seems to be related to GitHub and https, I have not modified any other DietPi scripts to use curl yet, however, I will keep an eye on user feedback to see if this issue requires further coding.

closed 2015-12-01 by Fourdee