MichaIng/DietPi #126: Build instructions

platy, 2015-11-28:
Do you have any build instructions? I'd like to make an image optimised for RPi headless embedded deployment using docker containers and DietPi looks like the best place to start.

Fourdee, 2015-11-28:
Adding DietPi to a Debian system is a manual process. I do have notes, which I follow and your are you welcome to try.

These notes apply to Jessie.
```
#-------------------------------------------------------------------------
//DietPi on a fresh install
#-------------------------------------------------------------------------

#Jessie , unified apt sources.
rm /etc/apt/sources.list.d/deb-multimedia.list
# ------- NOT RPI -------------------------
cat << _EOF_ > /etc/apt/sources.list
deb http://ftp.debian.org/debian jessie main contrib non-free
deb http://ftp.debian.org/debian jessie-updates main contrib non-free
deb http://security.debian.org jessie/updates main contrib non-free
deb http://ftp.debian.org/debian jessie-backports main contrib non-free
_EOF_
# ------- NOT RPI -------------------------

#Remove following Jessie
apt-get purge -y rsyslog logrotate man-db manpages vim vim-common vim-runtime vim-tiny mc mc-data
apt-get autoremove --purge -y

#install
echo -e "CONF_SWAPSIZE=100" > /etc/dphys-swapfile
apt-get install bzip2 psmisc bash-completion cron whiptail sudo ntp ntfs-3g dosfstools parted hdparm pciutils usbutils zip htop wput wget fake-hwclock dphys-swapfile curl unzip ca-certificates console-setup console-data console-common keyboard-configuration wireless-tools --no-install-recommends

#firmware
apt-get install firmware-realtek firmware-ralink firmware-brcm80211 firmware-atheros -y --no-install-recommends

#DIETPI STUFF
#Create folders
mkdir /DietPi
mkdir -p /mnt/usb_1
mkdir -p /mnt/samba
mkdir -p /mnt/ftp_client
echo -e "Samba client can be installed and setup by DietPi-Config.\nSimply run: dietpi-config 8" > /mnt/samba/readme.txt
echo -e "FTP client mount can be installed and setup by DietPi-Config.\nSimply run: dietpi-config 8" > /mnt/ftp_client/readme.txt

/boot/dietpi/dietpi-logclear 2

#FSTAB
cp /boot/dietpi/conf/fstab /etc/fstab

#setup dietpi service
echo 1 > /boot/dietpi/.install_stage
cp /boot/dietpi/conf/dietpi-service /etc/init.d/dietpi-service
chmod +x /etc/init.d/dietpi-service
update-rc.d dietpi-service defaults 00 80
service dietpi-service start

#Cron jobs
cp /DietPi/dietpi/conf/cron.daily_dietpi /etc/cron.daily/dietpi
chmod +x /etc/cron.daily/dietpi
cp /DietPi/dietpi/conf/cron.hourly_dietpi /etc/cron.hourly/dietpi
chmod +x /etc/cron.hourly/dietpi

#ntp
rm /etc/cron.daily/ntp &> /dev/null
rm /etc/init.d/ntp &> /dev/null

#/etc/sysctl.conf
echo -e "vm.swappiness=1" >> /etc/sysctl.conf
echo -e "net.ipv6.conf.all.disable_ipv6 = 1" >> /etc/sysctl.conf
echo -e "net.ipv6.conf.default.disable_ipv6 = 1" >> /etc/sysctl.conf
echo -e "net.ipv6.conf.lo.disable_ipv6 = 1" >> /etc/sysctl.conf

#Blacklist ipv6
echo -e "blacklist ipv6" > /etc/modprobe.d/blacklist-ipv6.conf

[... 118 lines cut ...]
_EOF_
systemctl enable rc-local.service
systemctl daemon-reload

#Shutdown SSH/Dropbear before reboot
cat << _EOF_ > /etc/systemd/system/kill-ssh-user-sessions-before-network.service
[Unit]
Description=Shutdown all ssh sessions before network
DefaultDependencies=no
Before=network.target shutdown.target

[Service]
Type=oneshot
ExecStart=/usr/bin/killall sshd && /usr/bin/killall dropbear

[Install]
WantedBy=poweroff.target halt.target reboot.target
_EOF_
systemctl enable kill-ssh-user-sessions-before-network
systemctl daemon-reload

dpkg-reconfigure tzdata
dpkg-reconfigure keyboard-configuration #Keyboard must be plugged in for this to work!
dpkg-reconfigure locales
```

blindpet, 2015-11-28:
I think @platy is interested in the build instructions for your raw debian base. I'd also love to see your script for building your debian base sans the DietPi goodies.

Fourdee, 2015-11-28:
As for making the image finalised for release:
```
/DietPi/dietpi/finalise
poweroff
```

platy, 2015-11-30:
This is what I was looking for, Thanks!

closed 2015-11-30 by platy

xenfomation, 2016-05-10:
For what it's worth - 99% of the instructions still work.  I was trying to get a VM prepped for you in the most extreme sense of the word "Diet"... Off to work!