MichaIng/DietPi #127: Clean install dietpi jessie on raspberry pi A+ results in black screen

bc547, 2015-11-29:
Hi,

I just downloaded DietPi_RPi-(Jessie).7z from dietpi.net and wrote the image file (DietPi_v94_RPi-(Jessie).img) to a 8GB sdcard using rufus (https://rufus.akeo.ie/). I did not modify any files on the sdcard afterwards. Next I put this sdcard in a raspberry pi A+.

At first boot, everything seems ok and there's a lot of messages scrolling on the screen. It automatically reboots 2 times. After the last boot, I can see the resize of the filesystem is happening and at the same time, systemd wants to restart. After a couple of seconds, systemd seems to force a restart. 

After these messages, the system does not boot anymore. Only the red led is on, no flickering at all of the green led and no hdmi output anymore. Even after a powercycle, this remains the case.

Installing another OS (minibian Jessie) is no problem and everything runs stable. Just in case, I also tried with 2 different brands of sdcards and 2 different rpi's and 2 different power supplies (at least 2A). 

Any advise to get a working dietpi installation?

bc547, 2015-11-29:
Seems a sdcard issue. 3th card seems to work.

closed 2015-11-29 by bc547

Fourdee, 2015-11-29:
Power issues can also cause SD card corruptions and similar issues. 
To be on the safe side, I'd suggest purchasing one of the Official RPI PSU's if you do not own one already.