MichaIng/DietPi #128: https://letsencrypt.org/ | free SSL certs

Fourdee, 2015-11-30:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=234

Look into https://letsencrypt.org/
Free SSL certs.

Fourdee, 2015-12-04:
> https://github.com/letsencrypt/letsencrypt

Gonna give it a spin :)

Fourdee, 2015-12-04:
Not getting anywhere with this:
```
wget https://github.com/letsencrypt/letsencrypt/archive/master.zip -O package.zip
unzip package.zip
rm package.zip
cd letsencrypt*
```
```#Create cert
./letsencrypt-auto certonly --agree-tos --standalone --email admin@dietpi.com -d dietpi.com```

![image](https://cloud.githubusercontent.com/assets/7077026/11596743/df6611a0-9aae-11e5-93a9-9b0c1493798a.png)

If I stop the apache2 service before running ./letsencrypt-auto (as it requests) I get:
```
Failed authorization procedure. dietpi.com (http-01): urn:acme:error:unauthorized :: The client lacks sufficient authorization :: Invalid response from http://dietpi.com/.well-known/acme-challenge/B4QKaRU_IEBetWWqmdaTBRmO1iOsrBZx-IGFAhWW7GE [46.101.57.99]: 404

IMPORTANT NOTES:
 - The following 'urn:acme:error:unauthorized' errors were reported by
   the server:

   Domains: dietpi.com
   Error: The client lacks sufficient authorization
```

Fourdee, 2015-12-04:
[image]

https://fuzon.co.uk/dietpi/index.html

```
wget https://github.com/letsencrypt/letsencrypt/archive/master.zip -O package.zip
unzip package.zip
rm package.zip
cd letsencrypt*
./letsencrypt-auto
```

R1CH-T, 2015-12-04:
Thanks for persevering.
I got the email updates from GitHub and had a feeling that you'd put in too many command line options on your first tries.
I'll give it a go myself tomorrow, after work.

Thanks for trying this out.

thechildofroth, 2015-12-04:
Well done Dan, looking forward to seeing this continue to develop.

Fourdee, 2015-12-04:
@R1CH-T Yep, had one of those "moments" :)

Fourdee, 2015-12-04:
@thechildofroth I can't take any credit, Letsencrypt is simply awesome. 5 lines of code to get a free SSL cert and automatic setup for Apache2.

They really nailed this, exactly what the Linux world needed.

thechildofroth, 2015-12-04:
Roll on DP103! ;-)

Remember, it's a 90 day cert, are you putting in the cron jobs to make sure
they're updated in a timely fashion?

Fourdee, 2015-12-05:
Interesting only 90 days, never knew.

Yep cron job might work, depends if we can complete the renewal without restarting the webserver (no downtime). I'll have a look.

Date: Fri, 4 Dec 2015 13:57:40 -0800

Fourdee, 2015-12-05:
```
An unexpected error occurred:
There were too many requests of a given type :: Error creating new cert :: Too m                                          any certificates already issued for: ddns.net
```
Well there goes my testing for the day. Seems my IP is being blocked from creating any new certs lol.

R1CH-T, 2015-12-06:
>
    There were too many requests of a given type :: Error creating new cert :: Too many certificates already issued for: ddns.net
>Well there goes my testing for the day. Seems my IP is being blocked from creating any new certs lol.

Interesting.
I notice that your test page is now down. When I visited https://fuzon.co.uk/dietpi/index.html on Friday, I was able  to get an encrypted version of all sub-pages, i.e. https://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=234&p=970#p970
Now, none of it is encrypted. Did you back up the certificates and the config files which Let's Encrypt modified?
Backing this sort of stuff up is usually a good idea. Does the Let's Encrypt client allow for re-installing already-issued certificates?
Is this why you are getting the above error message?
It seems to be telling you that it won't issue a certificate for **ddns.net** and not **fuzon.co.uk**, which is your intended target domain.
Obviously **ddns.net** seems to be your dns provider for this site, but I don't see why this should affect the certificate in any way. You're not using a sub-domain of **ddns.net** like **fuzon.ddns.net** or anything like that. (Back when *dyndns* used to provide their free service, I used to have such a site address myself.)
If it helps, as far as I know and for the moment at least, Let's Encrypt doesn't provide *wildcard* certificates, so any sub-domains (*.mysite.com), i.e. blog.fuzon.co.uk will need to be applied for individually.
Strange indeed.
:/

Fourdee, 2015-12-06:
@R1CH-T 
As http://fuzon.co.uk is hosted on my home network (vm webserver), I changed the port forwarding for 443 to my RPi for testing. I was able to generate certs at first.
I generated a NoIp address to test after fuzon.co.uk was being denied. Feels like an IP block to prevent abuse.

Heres my notes before being denied further certs:
```
wget https://github.com/letsencrypt/letsencrypt/archive/master.zip -O package.zip
unzip package.zip -d /root
rm package.zip

letsencrypt_domain="fuzon.co.uk"
letsencrypt_email="admin@fuzon.co.uk"
letsencrypy_keysize=2048

#/etc/apache2/sites-available/000-default.conf - add 
#ServerName "$letsencrypt_domain"

service apache2 restart

cd /root/letsencrypt*
./letsencrypt-auto --duplicate --agree-tos --no-redirect --rsa-key-size $letsencrypy_keysize --email $letsencrypt_email -d $letsencrypt_domain
```
The plan is to store these settings for the user and integrate in a whiptail frontend. This way we can use cron to renew.

I'll have another play today.

>Backing this sort of stuff up is usually a good idea. Does the Let's Encrypt client allow for re-installing already-issued certificates?

Not tested this yet.

R1CH-T, 2015-12-06:
I've had guests this weekend, so I haven't had a go at this yet.
Sorry, will test ASAP.
I'm particularly interested in Nginx integration, so I'll prioritise that.
Thanks,
Rich.

Fourdee, 2015-12-06:
@R1CH-T I'am working on LetsEncrypt for apache2 integration on a DietPi system, for now. I'll link the script when it's done. Would you be up for assisting with the Nginx side?

Fourdee, 2015-12-06:
Little preview of frontend:
[image]

Fourdee, 2015-12-06:
> https://github.com/letsencrypt/letsencrypt/issues/1721#issuecomment-161955087

Turns out theres a hard limit to cert requests per week. No more testing for me then :-1: . Any chance I could get one of you guys to help test on your system?

thechildofroth, 2015-12-06:
> letsencrypt/letsencrypt#1721 (comment)
>
> Turns out theres a hard limit to cert requests per week. No more testing
> for me then [image: :-1:] . Any chance I could get one of you guys to
> help test on your system?

More than happy to but I'll need some guidance - I did actually get to play
the day before the public launch as I'd put it of the private beta (late)
and got the app installed etc. Problem is I lack webserver admin skills so
was somewhat behind the curve once I was installed.

Fourdee, 2015-12-07:
@thechildofroth Thanks for offering your assistance.

I can walk you through the steps needed. As this will be testing, do you have a spare RPi/card to use for this purpose?

Are you able to:
- Setup a domain name (eg: noip) and point it to your device?
- Forward port 80 and 443 TCP to your device?

thechildofroth, 2015-12-07:
yup, I actually have my own domain (part of why I was interested in getting
an ssl cert). I'm ok with the router setup etc and have had it working with
my domain and things like owncloud etc) it's the ssh configuration side of
things I'm less sure about.

I'd also like to (potentially) migrate my Subsonic install to https (and
that's up and running already) so perhaps we can start with that one (if I
move it from subdomain.subsonic.org to mydomain.co,uk)

Fourdee, 2015-12-07:
Brilliant.
For now, this will be a single test on a fresh install of DietPi. I just want to get a gauge if the new script works. I would wait until v103 is released before applying this to any active systems.
Ok heres what you will need to do:- Write a fresh DietPi image.- Open up /boot/dietpi.txt and change the gitbranch to testing:
gitbranch=testing- Plug in and power on.
- Go through the normal DietPi 1st run setup- When DietPi-Software appears, select the following for installation:LAMPLetsEncrypt
- Wait for installation to finish and system to reboot. Be patient, letsencrypt takes a while.
- Ensure you have tcp port 80 and 443 forwarded to your device.
- run the following:/DietPi/dietpi/dietpi-letsencrypt
- Enter your email address and domain name. Then select "Apply".
Let me know what happens :)

Date: Mon, 7 Dec 2015 10:15:38 -0800
From: notifications@github.com
To: DietPi@noreply.github.com
CC: daniel_haze@hotmail.com
Subject: Re: [DietPi] https://letsencrypt.org/ | free SSL certs (#128)

yup, I actually have my own domain (part of why I was interested in getting

an ssl cert). I'm ok with the router setup etc and have had it working with

my domain and things like owncloud etc) it's the ssh configuration side of

things I'm less sure about.

I'd also like to (potentially) migrate my Subsonic install to https (and

that's up and running already) so perhaps we can start with that one (if I

move it from subdomain.subsonic.org to mydomain.co,uk)

thechildofroth, 2015-12-07:
Hi Dan,

Thanks for the easy to follow instructions, happy to confirm this as
working. I can get to the apache/debian 'it's working' index page and have
a lovely green https in the address bar, akso when I click the padlock it
says nice things about what a great certificate I have etc etc.

I'm going to throw owncloud in to the mix and see what happens.

thechildofroth, 2015-12-08:
I left owncloud installing when I went to bed last night, took a reboot
when I woke up this morning but after that it's all working over https.

Forgot to mention last night, in the letsencrypt settings I had to list my
domain with the www to make it work (although this might be a failing of
the way I've got my dns settings?) I set redirect to 'enable' and auto
renew to 'enable' too (but obviously it will take a while to see if that
works!), key size was set to 2048.

Fourdee, 2015-12-08:
@thechildofroth Awesome! Thanks for testing, really appreciate it. :+1: 

I'll add the www. to the default entry (eg: www.mydomain.com). Could be a dns issue, but adding www. shouldn't do any harm.

Without your testing, this wouldn't of made it to v103. So hats off :)

thechildofroth, 2015-12-08:
Glad to be of assistance.

R1CH-T, 2015-12-08:
Hi.
Sorry that I've been dragging my feet on this, but I haven't had the chance to look in here, let alone have a go at this.
Anyway: Great work, guys; you seem to be getting somewhere!

>@R1CH-T I'am working on LetsEncrypt for apache2 integration on a DietPi system, for now. I'll link the script when it's done. Would you be up for assisting with the Nginx side?

Short answer: Yes, of course.
Long answer: I've just been experimenting with getting this to work with Nginx on an Ubuntu 14.04 VM, but keep getting errors and polite refusals from the Lets Encrypt installer so far.
They do state that Nginx support is experimental at this stage and in reading through the installation scripts, I've forund many sections where it has been disabled.
I'll keep plugging away in the little bits of time available to me when and where I can and will report back as I go along.

>@thechildofroth Said: Forgot to mention last night, in the letsencrypt settings I had to list my domain with the www to make it work (although this might be a failing of the way I've got my dns settings?)

>@Fourdee Said: I'll add the www. to the default entry (eg: www.mydomain.com). Could be a dns issue, but adding www. shouldn't do any harm.

Well done, and FYI:
The "www" thing would seem to be a peculiarity of the system, I think.
I encountered this kind of thing when I was researching this stuff a year ago.
I have all sorts of notes, links and bookmarks stored away and one of the more useful ones I found at the time was this:
http://www.troyhunt.com/2013/09/the-complete-guide-to-loading-free-ssl.html
Under the section "4. Request an SSL cert" it illustrates the traditional process around certifying domains and sub-domains and gives you an idea of what the various classes of certificates are. It was instrumental in my getting this stuff to work the first time around.
I ended up by getting certificates for various sub-domains such as **www. blog. cloud. mail.** etc. so I had a different sub-domain for each sort of application.
I think that the **www.** is the *front page* url or a *catch-all* for the entire domain, if you see what I mean.

Anyway, I'm out of time now so I'll post again ASAP
Thanks again, guys.
Rich.

Fourdee, 2015-12-09:
> They do state that Nginx support is experimental at this stage and in reading through the installation scripts, I've forund many sections where it has been disabled.
Yep i also seen that. Theres no rush for this, probably best to wait until Nginx support is more stable.

> The "www" thing would seem to be a peculiarity of the system, I think.
I encountered this kind of thing when I was researching this stuff a year ago.

Thanks for sharing your experiences with this. 
I didn't require the www. part for my fuzon.co.uk site, but i'am using NoIp Plus managed. Maybe the free NoIp only supports www.* forwarding?
@thechildofroth : How is your domain routed to your home network (eg: NoIp Free)?

thechildofroth, 2015-12-09:
I own the domain and have access to the DNS table (it's hosted with Mythic
Beasts as they also provide an api that means I can use curl to update the
IP in the DNS record allowing me to run (for instance) a mail server on a
domestic (dynamically allocated) IP).

As I mentioned before though I'm a numpty with webserver admin so I may
have done it wrong!

R1CH-T, 2015-12-09:
@Fourdee Said:
>Thanks for sharing your experiences with this.
No problem, I need to look at this stuff again anyway.
I'm paying for a domain name and have had little use of it so far, apart from the enjoyable learning experience I had in the beginning in setting up various web services.
I certainly haven't got anything set up which compares to your growing DietPi Empire! :)

>I didn't require the www. part for my fuzon.co.uk site, but i'am using NoIp Plus managed. Maybe the free NoIp only supports www.* forwarding?
What I was trying to get at above is that the **www.** sub-domain *isn't treated as a sub-domain*, as far as I can fathom. It seems to be *synonymous with the plain http address*, so http://www.dietpi.com is treated *or should be treated* the same as http://dietpi.com
I may be wildly wrong about this, but I think that the **www.** part is protected, but not classed as a sub-domain in the same way as any others you may create (ie **blog.** etc.).
As I say, I'm definitely not an expert in this, so *your mileage may vary*.

R1CH-T, 2015-12-09:
@thechildofroth Said:
>I own the domain and have access to the DNS table (it's hosted with Mythic Beasts as they also provide an api that means I can use curl to update the IP in the DNS record...

I got my DNS record to update via a Perl script, modified from https://calomel.org/dynamic_dns_ddns.html
(They now provide a windows *.exe* to do this job, but this is no good for me.)

>...allowing me to run (for instance) a mail server on a domestic (dynamically allocated) IP).

I actually managed to set up a POP and IMAP Mail Server (with SSL) on a local VM, got the thing to work, using openSMTPD and Dovecot and managed to get it up and running on the internet via NameCheap. I didn't get an SQL database set up, but instead used the Unix Pluggable Authentication Module (PAM) to create the email accounts on my test machine.
This was all done during a marathon hacking session a year ago, but I'm going to have to relearn it now and probably rewrite my notes on the subject as some of this stuff has changed...

>**Nope.**
Not at all. Unless you do this sort of stuff as your daily job, it will *always* get the better of you.

Fourdee, 2015-12-10:
Available as a optimized installation in DietP-Software, with DietPi v103.
Info: http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&p=1061#p1062

closed 2015-12-10 by Fourdee