MichaIng/DietPi #129: DietPi-Software | Add Domoticz installation option

Fourdee, 2015-11-30:
Home automation software:

> http://domoticz.com/
> http://fuzon.co.uk/phpbb/viewtopic.php?f=12&t=222&sid=744cfa3150949e4c65d4df42ae4f7500#p863
> https://www.domoticz.com/wiki/Linux

mlamie, 2017-01-31:
Hi Fourdee, can you include Domoticz to the DietPi software list? They have improved there installtion method and now is fully installed with 1 command. I already tested it and works great on DietPi.

`sudo curl -L install.domoticz.com | sudo bash`

See also: https://www.domoticz.com/wiki/Installing_and_running_Domoticz_on_a_Raspberry_PI

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

vikasjee, 2018-06-01:
Need to run domoticz on dietpi. Please help

elRadix, 2018-06-01:
You could try to follow this guide. 
https://www.domoticz.com/wiki/Linux

I have tried it once and got it working but I switched to Home-Assistant which is much better - personal opinion.

MichaIng, 2019-01-30:
@boschr and also the others:
You may want to add this request to our FeatHub page: https://feathub.com/Fourdee/DietPi/
Link this topic there and collect votes. The best way we can set priorities on adding requests 😉.

boschr, 2019-01-31:
@MichaIng i will do! Collecting votes will not be a problem i guess.

MichaIng, 2019-02-12:
Reached 2nd place on FeatHub: https://feathub.com/Fourdee/DietPi/+37
Will reopen.

ilgrank, 2019-09-07:
Now 1st place on FeatHub :)
Thanks guys!

MichaIng, 2020-03-08:
Install test via official install script:
```
2020-03-08 21:30:22 root@VM-Buster:~# curl -sSL install.domoticz.com | bash
:::
::: You are root.
::: Verifying free disk space...
stat: cannot stat '/var/lib/apt/lists/': No such file or directory
2020-03-08 21:30:32 root@VM-Buster:~# apt update
Get:1 https://deb.debian.org/debian buster InRelease [122 kB]
Get:2 https://deb.debian.org/debian buster-updates InRelease [49.3 kB]
Get:3 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Get:4 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Get:5 https://deb.debian.org/debian buster/main amd64 Packages [7,907 kB]
Get:6 https://deb.debian.org/debian buster/contrib amd64 Packages [51.0 kB]
Get:7 https://deb.debian.org/debian buster/non-free amd64 Packages [88.0 kB]
Get:8 https://deb.debian.org/debian buster-updates/main amd64 Packages [7,380 B]
Get:9 https://deb.debian.org/debian-security buster/updates/main amd64 Packages [181 kB]
Get:10 https://deb.debian.org/debian-security buster/updates/non-free amd64 Packages [552 B]
Get:11 https://deb.debian.org/debian buster-backports/contrib amd64 Packages [5,204 B]
Get:12 https://deb.debian.org/debian buster-backports/non-free amd64 Packages [19.5 kB]
Get:13 https://deb.debian.org/debian buster-backports/main amd64 Packages [225 kB]
Fetched 8,767 kB in 7s (1,186 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
2020-03-08 21:31:05 root@VM-Buster:~# ln -s /tmp/apt/
lists/           pkgcache.bin     srcpkgcache.bin
2020-03-08 21:31:05 root@VM-Buster:~# ln -s /tmp/apt/lists /var/lib/apt/lists
2020-03-08 21:31:56 root@VM-Buster:~# curl -sSL install.domoticz.com | bash
:::
::: You are root.
::: Verifying free disk space...
:::
::: Checking apt-get for upgraded packages.... done!
:::
::: Your system is up to date! Continuing with Domoticz installation...
:::    Checking for apt-utils... installed!
:::    Checking for whiptail... installed!
:::    Checking for git... installed!
:::    Checking for curl... installed!
:::    Checking for unzip... installed!
:::    Checking for wget... installed!
:::    Checking for sudo... installed!
:::    Checking for cron... installed!
:::    Checking for libudev-dev... installed!
:::    Checking for libcurl4... installed!
:::    Checking for libusb-0.1... installed!
::: Downloading Domoticz
::: Current User: root
HTTP Port:  8080
HTTPS Port:  443
:::
::: Destination folder=/mnt/dietpi_userdata/domoticz
...
```
🈴 Does not support custom APT lists dir, but this can be contributed easily, workaround as done via symlink works
🈺 Installer is interactive, install dir can be chosen, HTTP and/or HTTPS can be selected.
🈴 Installed sysvinit service fails on Stretch+ where libssl1.0.0 library is not available:
```
Mar 08 21:48:05 VM-Buster domoticz.sh[1233]: /mnt/dietpi_userdata/domoticz/domoticz: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory
```
- I guess for using current libssl1.1, binaries would need to be recompiled 🤔.

🈺 Luckily DietPi-Software ships `libssl1.0.0` as backwards-compatibility package, since there were some more software titles with the same issue.
```
2020-03-08 21:56:32 root@VM-Buster:~# dietpi-software install 126
...
Mar 08 22:02:44 VM-Buster systemd[1]: Starting LSB: Home Automation System...
Mar 08 22:02:44 VM-Buster domoticz.sh[2203]: 2020-03-08 22:02:44.648  Status: Domoticz V4.10717 (c)2012-2019 GizMoCuz
Mar 08 22:02:44 VM-Buster domoticz.sh[2203]: 2020-03-08 22:02:44.648  Status: Build Hash: b38b49e5, Date: 2019-05-09 13:04:08
Mar 08 22:02:44 VM-Buster domoticz.sh[2203]: 2020-03-08 22:02:44.648  Status: Startup Path: /mnt/dietpi_userdata/domoticz/
Mar 08 22:02:44 VM-Buster domoticz.sh[2203]: domoticz: Domoticz is starting up....
Mar 08 22:02:44 VM-Buster domoticz[2208]: Domoticz is starting up....
Mar 08 22:02:44 VM-Buster domoticz[2210]: Domoticz running...
Mar 08 22:02:44 VM-Buster systemd[1]: Started LSB: Home Automation System.
```
- Both, HTTP and HTTPS work fine for web UI access (besides default cert is not trusted of course).

Okay while this works generally, it shows there there is quite some update required for the official installer, especially a systemd unit would be nice.
For now I would skip the installer and implement it manually, which is not very hard indeed:
🈯️ Pre-compiled binaries for all architectures we support: https://www.domoticz.com/downloads/
- Great is that they use the same `uname -m`-based naming scheme + ARMv6 is supported by the armv7l binaries obviously (the installer replaces armv6l by armv7l before starting the download).
- Issue with libssl1.0.0 persists, for now we can install our workaround package, and when someone finds time build it for modern libraries.

🈯️ systemd unit: https://www.domoticz.com/wiki/Linux

Sadly there seem to be no native way to separate install and configs? However minor issue, reinstalls will just follow the internal updater and merge new files with the little downside that old/obsolete files will stay in place.

MichaIng, 2020-03-09:
PR up, but not yet ready for testing: #3415

MichaIng, 2020-03-10:
Okay PR now ready for testing. If one wants to try (Do a `dietpi-backup` first since this includes DietPi update to development version!):
```
dietpi-backup 1
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=dev' /DietPi/dietpi.txt
dietpi-update 1
dietpi-software install 140
```

Joulinar, 2020-03-13:
@MichaIng 
I was testing installation on RPi3B+ as well as VM and got the following error message during installation

```
[  OK  ] DietPi-Software | Comment in /mnt/dietpi_userdata/domoticz/domoticz.conf converted to setting: userdata_path=/mnt/dietpi_userdata/domoticz
useradd: group 'dealout' does not exist
chown: invalid user: ‘domoticz:domoticz’
```

This leads Domoticz to fail during start

```
root@DietPiVM1:~# systemctl status domoticz.service
● domoticz.service - Domoticz (DietPi)
   Loaded: loaded (/etc/systemd/system/domoticz.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-03-13 11:48:54 CET; 2min 25s ago
  Process: 583 ExecStart=/opt/domoticz/domoticz -f /mnt/dietpi_userdata/domoticz/domoticz.conf (code=exited,
status=217/USER)
 Main PID: 583 (code=exited, status=217/USER)

Mar 13 11:48:54 DietPiVM1 systemd[1]: Started Domoticz (DietPi).
Mar 13 11:48:54 DietPiVM1 systemd[583]: domoticz.service: Failed to determine user credential
s: No such process
Mar 13 11:48:54 DietPiVM1 systemd[583]: domoticz.service: Failed at step USER spawning /opt/d
omoticz/domoticz: No such process
Mar 13 11:48:54 DietPiVM1 systemd[1]: domoticz.service: Main process exited, code=exited, sta
tus=217/USER
Mar 13 11:48:54 DietPiVM1 systemd[1]: domoticz.service: Failed with result 'exit-code'.
root@DietPiVM1:~#
```

MichaIng, 2020-03-13:
@Joulinar 
Many thanks for testing. Lol dumb typo by me, it's `dialout`. Quick addition after I ran all tests already.

MichaIng, 2020-03-13:
Fixed: https://github.com/MichaIng/DietPi/commit/d05d611e8c85201fe93ea29602d69cc520848e32

Joulinar, 2020-03-13:
ok looks fine now

closed 2020-05-03 by MichaIng