MichaIng/DietPi #133: ram-log always makes a directory as root owner at boot and does not inherent original owner attribute.

epgdatacapbon, 2015-12-06:
Some user program (own build software) makes log directories as other than root.
However, ram-log always makes such directories in ramdisk as root owner at every boot.
Thus, the program cannot create any file in the directories (without permission).

Fourdee, 2015-12-06:
I have plans to upgrade DietPi-Ramlog to use Rsync for v103. This would keep file attributes during creation and shutdown.

In the mean time, you can either run your program as root, or, change the DietPi logging mode to "Full". This will disable DietPi-Ramlog: http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&start=20#p68

Fourdee, 2015-12-06:
Resolved for v103.

DietPi-Ramlog will now preserve file ownerships.
Turns out rysnc isn't needed. Amazing what you can do with cp -p --attributes-only.

closed 2015-12-06 by Fourdee

epgdatacapbon, 2015-12-06:
Great. Thanks.