MichaIng/DietPi #134: DietPi-Software | Add LEPP

keiser1080, 2015-12-06:
Hi,

new software suggestion Nginx   / postgrees / Php.
Why segregation of postgresql :)
posgresql has lower memory footprint.

Let my know if i can help givin installation process of testing the script.

Fourdee, 2015-12-06:
Hi Keiser,

Unfortunately, I do not have the time to look into this at the moment. If you have a working installation method for this stack, by all means let me know and I'll see if we can get it added into DietPi.

xenfomation, 2016-05-13:
@Fourdee and @keiser1080 --

Since I have been working with the webstacks recently, I will take a look at the process this week (manually) and follow up to see about converting it to an installable option.

Cheers!

keiser1080, 2016-05-13:
If you can wait i can sens a bash script semi automates next week, si you
can addapt thé script?

xenfomation, 2016-05-13:
Sure!

Fourdee, 2016-07-23:
Outdated and no progress, marking as closed. Please reopen if required.

closed 2016-07-23 by Fourdee