MichaIng/DietPi #137: DietPi-Software | Add LibreNMS installation option

Fourdee, 2015-12-07:
> https://twitter.com/ewan2395/status/673606407808270337

Install:
> http://docs.librenms.org/Installation/Installation-(Debian-Ubuntu)/

f0o, 2015-12-20:
Hi!
just checking what the status is and if you need any assistance :)

Cheers!

Fourdee, 2015-12-20:
@f0o Just finishing up some automation features, then i'll start on this. I'll give you a buzz if I get stuck :+1:

f0o, 2015-12-20:
awesome! just give me or @LibreNMS a shout! :+1:

Fourdee, 2015-12-20:
@f0o Hi Daniel, having a few issues with the install.

The end goal is to have librenms hosted in its directory location eg: http://server/librenms
Ideally in /var/www/librenms as DietPi's LAMP stack is shared with other www software (eg: owncloud/ympd etc)

Using http://docs.librenms.org/Installation/Installation-(Debian-Ubuntu)/ tutorial, heres what I have so far for a Jessie install:
```
#REQUIRES LAMP
#REQUIRES RSYSLOG
#DietPi will install these prereqs automatically when Librenms is selected.

#Librenms additional prereqs
apt-get install -y libapache2-mod-php5 php5-cli php5-mysql php5-gd php5-snmp php-pear php5-curl snmp graphviz php5-mcrypt php5-json apache2 fping imagemagick whois mtr-tiny nmap python-mysqldb snmpd php-net-ipv4 php-net-ipv6 rrdtool

#Create SQL DB
/DietPi/dietpi/func/create_mysql_db librenms dietpi root dietpi

#Grab librenms
wget https://github.com/librenms/librenms/archive/master.zip -O package.zip
unzip package.zip
rm package.zip

#Move contents to /opt/librenms
mkdir -p /opt/librenms
mv librenms-master/* /opt/librenms/
rm -R librenms-master

#Create librenms-user (is this really necessary? We could use the existing www-data user)
useradd librenms -d /opt/librenms -M -r
usermod -a -G librenms www-data

#Setup dir's and permissions
mkdir -p /opt/librenms/rrd
mkdir -p /opt/librenms/logs

chown -R librenms:librenms /opt/librenms
chown -R www-data:www-data /opt/librenms/logs
chmod -R 775 /opt/librenms/rrd

#Apache2 site conf
cat << _EOF_ > /etc/apache2/sites-available/librenms.conf
<VirtualHost *:80>
  DocumentRoot /opt/librenms/html/
  CustomLog /opt/librenms/logs/access_log combined
  ErrorLog /opt/librenms/logs/error_log
  AllowEncodedSlashes NoDecode
  <Directory "/opt/librenms/html/">
    AllowOverride All
    Options FollowSymLinks MultiViews
  </Directory>
</VirtualHost>
_EOF_

#snmpd minimal cfg
echo -e "rocommunity public 127.0.0.1" >> /etc/snmp/snmpd.conf

#enable mcrypt
php5enmod mcrypt

#Enable rewrite engine and librenms site
a2ensite librenms.conf
a2enmod rewrite
service apache2 restart
```
The issues are:
- 404 error with http://server/install.php

So I created a symlink | ln -sf /opt/librenms/html /var/www/librenms
- Starts ok. Then unable to create config.php error in install.php

I tend not to use /etc/apache2/sites-available/site.conf for DietPi. We throw the website into /var/www/program_name etc. So probably my lack of site.conf's knowledge is at fault :)

I'll try some more tomorrow, but if you can pickup on anything, i appreciate the advice.

f0o, 2015-12-20:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

No problem, I'll pop you an apache config + modifications needed to get it running in a subdirectory tomorrow.
If you can predict the sql creds you can even ship a premade config for libre :)

Cheers,
f0o

-----BEGIN PGP SIGNATURE-----

iQJPBAEBCgA5MhxEYW5pZWwgUHJldXNza2VyIChNb2JpbGVLZXkjMikgPGYwb0Bs
aWJyZW5tcy5vcmc+BQJWdx3TAAoJEJMZKtVWOFzBC78P/2V1sZrP2IesmB5DglEb
pOOw/eTRQW+BjFIhAbFLviluwuJqsbSsZO8nQHWJd5DQJO86h5DMRt5OSdFXuBk3
oAZVA93JpO5y458DYjXaTLH4hiO0ioR4vb2q6vEeHnXmXR6BONSLMEHbdfZ/VcPe
C+MNFVDwbfU/iZudlDn1jZaarM1rX7aKQbBwOXRlO4Jso/iL+djblc72Lg8kJira
tkyDPjOoZCCwrC2S8CRUR0X1RXqsgJpRN4dO4QMcr4Fa7h9iDh+oe2p21M8VRTk2
bd2p7Wp14OtATqWqt+4ErkHtKLqTyuQQ2hVoycAu8OdobY1kCLWKt8mbCqD/jCNY
m92K29VTgikQjuhVsPyQsNOVnksDvKp1UZSHvHk1xcU3Gn3COyZi2NGi+cxRLyhU
RcxRA0mtQyI9BcBG3sbJMxbakEunzNUHLpWF0YVVLYCfAEL92HfAHXbV1sAu0rce
D7vcgDatDf6LHWIhC7j1f5b5w/Tk8lni2O8pCU4+8k83bLwueujGRLAUpiXaRzS4
bYtXaPKrOWbc1/sodLMk0NsBxRk9brBuPXgo8/GOrX5PlQa5//eu0Oq4YER2med8
sFffql+09wZH5NNk/Wt+5q5fKaYPH9vCkpROEEcNkCx/v2BvDqGbUwF66B6lL9Qa
bTXap6TiG+MRDRmV6KvdKCxt
=seky
-----END PGP SIGNATURE-----

Fourdee, 2015-12-22:
@f0o 
```
No problem, I'll pop you an apache config + modifications needed to get it running in a subdirectory tomorrow.
```
Thanks Daniel, appreciate it.

```
If you can predict the sql creds you can even ship a premade config for libre :)
```
Awesome :+1: 
I'll also look into exporting the SQL DB and importing during install (as per Ampache install). The more DietPi can automate LIbrenms, the better for the end user. :)

f0o, 2015-12-23:
Terribly sorry for my delay.
X-Mas got me in panic mode haha

Anyways, have a look at shipping our releases instead of the master branch.
For sub-dirs you need to adjust the .htaccess file inside `html/`, modify the RewriteBase to suit your needs.

Generally spoken you should consider nginx+fpm instead of apache, but we of course support both so this advice is very tainted by personal opinion ;)

Also if possible suggest rrdcached to your users, it drops disk I/O by almost 40%!

I'm currently on the airplane towards `home` but will compile you the changes needed in 2Hrs - Promised!

f0o, 2015-12-24:
It seems you really only need to adjust https://github.com/librenms/librenms/blob/master/html/.htaccess and fix the `RewriteBase /` to `/librenms/`

Fourdee, 2015-12-24:
@f0o Xmas panic here also hehe :)

Thanks for the info, appreciate it :+1: . I'll give it a spin as soon as I can and let you know how it goes. Santa may have to run a little late this year ;)

Fourdee, 2015-12-26:
@f0o 
Spent 4+ hours on this today testing multiple installations, methods, locations, configurations etc. 
LibreNMS will not function correctly, unless it is installed as the base / (and only) website on the system (replacing the 000-default).

The .htaccess changes have no effect. 
[image]

At a most basic level, it must be possible to install LibreNMS alongside additional websites? 
Maybe i'am doing something completely wrong (I don't know everything :) ). I'am using the installation guide http://docs.librenms.org/Installation/Installation-(Debian-Ubuntu)/ as a baseline,  however, i'am unable to find a way to achieve Librenms running anywhere except the base /var/www/ directory.

After installing LAMP, heres the current setup/installation i'am trying to achieve /var/www/librenms:
```
apt-get install -y libapache2-mod-php5 php5-cli php5-mysql php5-gd php5-snmp php-pear php5-curl snmp graphviz php5-mcrypt php5-json apache2 fping imagemagick whois mtr-tiny nmap python-mysqldb snmpd php-net-ipv4 php-net-ipv6 rrdtool

#Create SQL DB
/DietPi/dietpi/func/create_mysql_db librenms dietpi root dietpi

#download
wget https://github.com/librenms/librenms/archive/master.zip -O package.zip
unzip package.zip
rm package.zip

mkdir -p /opt/librenms
mv librenms-master/* /opt/librenms/
rm -R librenms-master

mkdir -p /opt/librenms
mv librenms-master/* /opt/librenms/

rm -R librenms-master

#Create librenms-user
useradd librenms -d /opt/librenms -M -r
usermod -a -G librenms www-data

#Setup dir's and permissions
mkdir -p /opt/librenms/rrd
mkdir -p /opt/librenms/logs

chown -R librenms:librenms /opt/librenms
chown -R www-data:www-data /opt/librenms/logs
chmod -R 775 /opt/librenms/rrd

# Add the following to /etc/apache2/sites-available/000-default
#AllowEncodedSlashes NoDecode
#<Directory "/var/www/librenms/html/">
#Require all granted
#AllowOverride All
#Options FollowSymLinks MultiViews
#</Directory>

#enable mcrypt
php5enmod mcrypt

#snmpd minimal cfg
echo -e "rocommunity public 127.0.0.1" >> /etc/snmp/snmpd.conf

#Enable rewrite engine
a2enmod rewrite
service apache2 restart

#symlink
ln -sf /opt/librenms/html /var/www/librenms

sed -i "/RewriteBase/c\RewriteBase /librenms/" /var/www/librenms/.htaccess
```

laf, 2016-01-28:
If you check the source code for the page, what do you see base href set to?

Fourdee, 2016-01-28:
@laf 
Since Librenms is your project, I assume this is some sort of hint? 

What page are you refering to https://github.com/librenms/librenms/tree/master/html ?

laf, 2016-01-28:
@Fourdee 

I'm just a dev :)

Sorry I meant the source code for the web page you sent a screenshot above (i.e the LibreNMS web interface).

Fourdee, 2016-01-28:
@laf 
Aha! :)
I'll setup a test installation tomorrow and take a look. Its been a while since I last tried it.

laf, 2016-01-28:
@Fourdee 

No problem. If it's remotely accessible feel free to give me a shout (I'm UK based). Do you have an irc channel?

Fourdee, 2016-01-29:
@laf No IRC channel at the moment.

# Ok, fresh install to /var/www/librenms 
- install.php works fine, no 404's
- index.php see image:
[image]

config.php:
```
$config['install_dir']  = "/var/www/librenms";
```

Will try a few things, but my HTML knowledge is limited. Although its just a path issue, must be an easy fix somewhere :)

f0o, 2016-01-29:
Perhaps it's as easy as adding the correct base_url?

laf, 2016-01-29:
Try setting $config['base_url'] = '/librenms/'; in config.php

laf, 2016-01-29:
lol, @f0o beat me to it.

Fourdee, 2016-01-29:
@f0o @laf 
```
$config['base_url']        = "/librenms/html/";
```
Resolved the issue :+1: 

Any chance we could get this implemented and applied when install.php is run? Seems required for non-base installations (eg: /var/www/librenms) where users run more than 1 website?

laf, 2016-01-29:
Just want to check that you haven't got the rest of our code accessible by librenms/includes/polling/functions.inc.php as an example?

Fourdee, 2016-01-30:
@laf 
Not 100%, but yes it looks like the poller isnt working. I'd assume everything in the parent directory of /librenms/html is not accessable, due to base_url ```/librenms/html/```.

Is there any tests I can run to verify this?

laf, 2016-01-30:
base_url ist just saying that all links would start with /librenms/html/. Nothing stopping you trying to go to /librenms/README.md

Give that a try?

closed 2017-12-30 by Fourdee