MichaIng/DietPi #139: Fully automated installation, with option to run user script via URL during 1st run.

Fourdee, 2015-12-08:
> https://www.raspberrypi.org/forums/viewtopic.php?p=856225#p856225

Fourdee, 2015-12-13:
Added following options to dietpi.txt:
```
#Automatically logs the root user in to start 1st run setup.
AUTO_AutoLogin=0

#Downloads and executes a custom script during 1st run setup. eg: http://myweb.com/script.sh | 0=disabled
AUTO_CustomScriptURL=0
```

Fourdee, 2015-12-13:
Added ability to choose which DietPi-Software will get automatically installed.
```
#Automated installs | 0 = normal, 1 = DietPi-Automation (use software list below) , 2 = grasshopper, 3 = Pi-hole
AUTO_Install_Index=0
# - Software to automatically install | 1=install | Requires AUTO_Install_Index=1
AUTO_DietpiSoftware_Install_DESKTOP_LXDE=0
AUTO_DietpiSoftware_Install_TIGHTVNCSERVER=0
AUTO_DietpiSoftware_Install_WEBSERVER_LAMP=0
AUTO_DietpiSoftware_Install_WEBSERVER_LASP=0
AUTO_DietpiSoftware_Install_WEBSERVER_LEMP=0
AUTO_DietpiSoftware_Install_WEBSERVER_LESP=0
AUTO_DietpiSoftware_Install_WEBSERVER_MYADMINPHP=0
AUTO_DietpiSoftware_Install_OWNCLOUD=0
AUTO_DietpiSoftware_Install_TRANSMISSION=0
AUTO_DietpiSoftware_Install_PHPBB=0
AUTO_DietpiSoftware_Install_HIFI=0
AUTO_DietpiSoftware_Install_KODI=0
AUTO_DietpiSoftware_Install_MINIDLNA=0
AUTO_DietpiSoftware_Install_OPENTYRIAN=0
AUTO_DietpiSoftware_Install_DIETPICAM=0
AUTO_DietpiSoftware_Install_DELUGE=0
AUTO_DietpiSoftware_Install_RASPCONTROL=0
AUTO_DietpiSoftware_Install_LINUXDASH=0
AUTO_DietpiSoftware_Install_PIHOLE=0
AUTO_DietpiSoftware_Install_SUBSONIC=0
AUTO_DietpiSoftware_Install_WEAVED=0
AUTO_DietpiSoftware_Install_WEBIOPI=0
AUTO_DietpiSoftware_Install_DIETPICLOUDSHELL=0
AUTO_DietpiSoftware_Install_HAPROXY=0
AUTO_DietpiSoftware_Install_SQUEEZEBOXSERVER=0
AUTO_DietpiSoftware_Install_WORDPRESS=0
AUTO_DietpiSoftware_Install_FAIL2BAN=0
AUTO_DietpiSoftware_Install_PHPSYSINFO=0
AUTO_DietpiSoftware_Install_PHPIMAGEGALLERY=0
AUTO_DietpiSoftware_Install_AMPACHE=0
AUTO_DietpiSoftware_Install_OPENVPNSERVER=0
AUTO_DietpiSoftware_Install_LETSENCRYPT=0
AUTO_DietpiSoftware_Install_WIRINGPI=0
AUTO_DietpiSoftware_Install_RPIGPIO=0
AUTO_DietpiSoftware_Install_RPII2C=0
AUTO_DietpiSoftware_Install_VIMFULL=0
AUTO_DietpiSoftware_Install_SSHCLIENT=0
AUTO_DietpiSoftware_Install_SMBCLIENT=0
AUTO_DietpiSoftware_Install_GITCLIENT=0
AUTO_DietpiSoftware_Install_BUILDESSENTIAL=0
AUTO_DietpiSoftware_Install_ALSABASE=0
AUTO_DietpiSoftware_Install_XSERVERXORG=0
AUTO_DietpiSoftware_Install_CURLFTPFS=0
AUTO_DietpiSoftware_Install_FFMPEG=0
AUTO_DietpiSoftware_Install_LOGGING_RSYSLOG=0
```

Fourdee, 2015-12-14:
> https://twitter.com/bookedirl/status/676180631689277441
Add automation support for locale tzdata keyboard-setup

Fourdee, 2015-12-15:
Added support for TZ/Locale/Keyboard layout:
```
#Language/Regional settings
# - Timezone eg: Europe/London America/New_York | Full list (TZ*): https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
AUTO_Timezone=Europe/London
# - Locale eg: en_GB.UTF-8 en_US.UTF-8 | Must be UTF-8 | Full list: cat /etc/locale.gen
AUTO_Locale=en_GB.UTF-8
# - Keyboard Layout eg: gb us de fr
AUTO_KeyboardLayout=gb
```

Fourdee, 2015-12-17:
+hostname

Fourdee, 2015-12-18:
```
#----------------------------------------------------------------
# D I E T - P I
# 1st run settings
#----------------------------------------------------------------
# All settings below will be applied on 1st boot/run.
#
# >> Networking Options -----------------------------------------
# - If both Ethernet and Wifi are enabled, Wifi will take priority and Ethernet will be disabled.

Ethernet_Enabled=1
Wifi_Enabled=0

# Enter your Wifi details below, if applicable (Case Sensitive).
Wifi_SSID=FuzonWifi
Wifi_KEY=MyAccessKey

# Enter your Static Network details below, if applicable.
Use_Static=0
Static_IP=192.168.0.100
Static_Mask=255.255.255.0
Static_Gateway=192.168.0.1
Static_DNS=8.8.8.8

# Hostname
Hostname=DietPi

# >> Automation Options -----------------------------------------
# - 1=Automatically logs the root user in after filesystem expansion to start 1st run setup. Set to 0 if you want to install over SSH.
AUTO_AutoLogin=0

#Automation Overrides for user inputs | Set these all to 1 if you want fully automated.
# - 1=Disable GPLv2 license information
AUTO_SkipLicensePrompt=0
# - 1=Reboot automatically once DietPi-Update has completed.
AUTO_DietpiSoftware_SkipUpdateRebootPrompt=0
# - 1=Disables question prompts.
AUTO_DietpiSoftware_SkipQuestions=0
# - 1=Disables prompt and ability to setup a dedicated USB drive for use on this system.
AUTO_DietpiSoftware_SkipUsbDrive=0
# - 0=Onscreen prompt if error occurs | 1=Ignore errors, no prompts. (logs errors to /root/DietPi-Automation.log)
AUTO_DietpiSoftware_IgnoreErrors=0

#Automated Software Installations | 0=Disabled, 1=DietPi-Automation (uses software list below) , 2=grasshopper, 3=Pi-hole
AUTO_Install_Index=0
#DietPi-Optimized Software to automatically install | 1=install | Requires AUTO_Install_Index=1
AUTO_DietpiSoftware_Install_DESKTOP_LXDE=0
AUTO_DietpiSoftware_Install_TIGHTVNCSERVER=0
AUTO_DietpiSoftware_Install_WEBSERVER_LAMP=0
AUTO_DietpiSoftware_Install_WEBSERVER_LASP=0
AUTO_DietpiSoftware_Install_WEBSERVER_LEMP=0
AUTO_DietpiSoftware_Install_WEBSERVER_LESP=0
AUTO_DietpiSoftware_Install_WEBSERVER_MYADMINPHP=0
AUTO_DietpiSoftware_Install_OWNCLOUD=0
AUTO_DietpiSoftware_Install_TRANSMISSION=0
AUTO_DietpiSoftware_Install_PHPBB=0
AUTO_DietpiSoftware_Install_HIFI=0
AUTO_DietpiSoftware_Install_KODI=0
AUTO_DietpiSoftware_Install_MINIDLNA=0
AUTO_DietpiSoftware_Install_OPENTYRIAN=0
AUTO_DietpiSoftware_Install_DIETPICAM=0
AUTO_DietpiSoftware_Install_DELUGE=0
AUTO_DietpiSoftware_Install_RASPCONTROL=0
AUTO_DietpiSoftware_Install_LINUXDASH=0
AUTO_DietpiSoftware_Install_PIHOLE=0
AUTO_DietpiSoftware_Install_SUBSONIC=0
AUTO_DietpiSoftware_Install_WEAVED=0
AUTO_DietpiSoftware_Install_WEBIOPI=0
AUTO_DietpiSoftware_Install_DIETPICLOUDSHELL=0
AUTO_DietpiSoftware_Install_HAPROXY=0
AUTO_DietpiSoftware_Install_SQUEEZEBOXSERVER=0
AUTO_DietpiSoftware_Install_WORDPRESS=0
AUTO_DietpiSoftware_Install_FAIL2BAN=0
AUTO_DietpiSoftware_Install_PHPSYSINFO=0
AUTO_DietpiSoftware_Install_PHPIMAGEGALLERY=0
AUTO_DietpiSoftware_Install_AMPACHE=0
AUTO_DietpiSoftware_Install_OPENVPNSERVER=0
AUTO_DietpiSoftware_Install_LETSENCRYPT=0
AUTO_DietpiSoftware_Install_WIRINGPI=0
AUTO_DietpiSoftware_Install_RPIGPIO=0
AUTO_DietpiSoftware_Install_RPII2C=0
#Linux Additional Software to automatically install | 1=install | Requires AUTO_Install_Index=1
AUTO_DietpiSoftware_Install_VIMFULL=0
AUTO_DietpiSoftware_Install_SSHCLIENT=0
# - If you want this to be auto mounted. Use DietPi-Config settings in this file to enter your details (SambaClient_*)
AUTO_DietpiSoftware_Install_SMBCLIENT=0
# - If you want this to be auto mounted. Use DietPi-Config settings in this file to enter your details (curlftpfs_*)
AUTO_DietpiSoftware_Install_CURLFTPFS=0
AUTO_DietpiSoftware_Install_GITCLIENT=0
AUTO_DietpiSoftware_Install_BUILDESSENTIAL=0
AUTO_DietpiSoftware_Install_ALSABASE=0
AUTO_DietpiSoftware_Install_XSERVERXORG=0
AUTO_DietpiSoftware_Install_FFMPEG=0
AUTO_DietpiSoftware_Install_LOGGING_RSYSLOG=0

#DietPi-Autostart | 0=Console 7=Console+auto root login | 1=Kodi 2=LXDE-Desktop 6=LXDE-VNCServer 5=DietPi-Cloudshell | Requires AUTO_Install_Index=1
AUTO_AutoStartTarget=0

#DietPi-Software Choice System
# -SSH Server Selection: 0=none , -1=dropbear , -2=opensshserver
AUTO_DietpiSoftware_SSHServerIndex=-1
# -File Server Selection: 0=none , -1=proftp , -2=samba
AUTO_DietpiSoftware_FileServerIndex=0
# -Logging Mode Selection: 0=none noclear, -1=ramlog 1h clear , -2=ramlog 1h save clear , -3=logrotate + rsyslog noclear
AUTO_DietpiSoftware_LoggingIndex=-1

#Language/Regional settings | Requires AUTO_Install_Index=1
# - Timezone eg: Europe/London America/New_York | Full list (TZ*): https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
AUTO_Timezone=Europe/London
# - Locale eg: en_GB en_US | Full list: cat /etc/locale.gen
AUTO_Locale=en_GB
# - Keyboard Layout eg: gb us de fr
AUTO_KeyboardLayout=gb

#Downloads and executes a custom script during last stage of 1st run setup. eg: http://myweb.com/script.sh | 0=disabled | Requires AUTO_Install_Index=1
AUTO_CustomScriptURL=0
```

Fourdee, 2015-12-21:
+option to format/setup dedicated USB drive.

Fourdee, 2015-12-21:
To prevent abuse of automation USB formatting, a file must exist on the 1st partition for this flag to be enabled:
```
/boot/Format_My_Usb_Drive
```

```
#Dedicated USB drive automation format (/dev/sda).
#	To use this feature, create a blank file: /boot/Format_My_Usb_Drive
#	If the file exists, the USB drive will be formatted to ext4. (WARNING: ALL DATA ON /dev/sda WILL BE DELETED. YOU HAVE BEEN WARNED!).
# - NB: Plug in 1 USB drive before powering on system. Use 1 USB drive ONLY.
```

Fourdee, 2015-12-22:
+
# >> Misc Options -----------------------------------
# - Size of dphys-swapfile to generate (MB) | 0=Disabled.
Swapfile_Size=100

Fourdee, 2015-12-23:
Modified Custom scripts to support local or online execution:
```
#Custom Script | Requires AUTO_Install_Index=1
# - Allows you to automatically execute a custom script at the end of DietPi installation.
# - Option 1 = Copy your script to /boot/Automation_Custom_Script.sh and it will be executed automatically.
# - Option 2 = Host your script online, then use AUTO_CustomScriptURL=http://myweb.com/myscript.sh , it will be downloaded and executed automatically. | 0=disabled
AUTO_CustomScriptURL=0
```

Fourdee, 2015-12-27:
Done: http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=273

closed 2015-12-27 by Fourdee

unclehowell, 2018-03-31:
Can we get this for version 6.4> ?  The auto login, static IP and disabling of prompts seem to have all come back in 6.4-6.5. What changed?

unclehowell, 2018-03-31:
This seems to work for me: 

AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=0
AUTO_SETUP_NET_USESTATIC=1
AUTO_SETUP_NET_STATIC_IP=192.168.87.10
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.87.1
AUTO_SETUP_NET_STATIC_DNS=10.0.0.1
AUTO_SETUP_NET_HOSTNAME=dietpi
AUTO_SETUP_NET_ETH_FORCE_SPEED=0
AUTO_SETUP_SWAPFILE_SIZE=0
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_GLOBAL_PASSWORD=dietpi
AUTO_SETUP_INSTALL_SOFTWARE_ID=60  # will install Wi-fi Hotspot 
AUTO_SETUP_SSH_SERVER_INDEX=-1
AUTO_SETUP_FILE_SERVER_INDEX=0
AUTO_SETUP_LOGGING_INDEX=-1
AUTO_SETUP_WEB_SERVER_INDEX=-2
AUTO_SETUP_AUTOSTART_TARGET_INDEX=0
AUTO_SETUP_TIMEZONE=Europe/London
AUTO_SETUP_LOCALE=en_GB.UTF-8
AUTO_SETUP_KEYBOARD_LAYOUT=gb
AUTO_SETUP_CUSTOM_SCRIPT_EXEC=0
CONFIG_HDMI_OUTPUT=1
CONFIG_CPU_GOVERNOR=ondemand
CONFIG_CPU_USAGE_THROTTLE_UP=50
CONFIG_CPU_MAX_FREQ=Disabled
CONFIG_CPU_MIN_FREQ=Disabled
CONFIG_CPU_ONDEMAND_SAMPLE_RATE=25000
CONFIG_CPU_ONDEMAND_SAMPLE_DOWNFACTOR=80
CONFIG_SMBCLIENT_COMPUTERNAME=computername
CONFIG_SMBCLIENT_SHARENAME=sharename
CONFIG_SMBCLIENT_USERNAME=username
CONFIG_SMBCLIENT_PASSWORD=password
CONFIG_CURLFTPFS_ADDRESS=192.168.0.100
CONFIG_CURLFTPFS_USERNAME=username
CONFIG_CURLFTPFS_PASSWORD=password
CONFIG_NFSCLIENT_ADDRESS=
CONFIG_PROXY_ENABLED=0
CONFIG_PROXY_ADDRESS=MyProxyServer.com
CONFIG_PROXY_PORT=8080
CONFIG_PROXY_USERNAME=
CONFIG_PROXY_PASSWORD=
CONFIG_BOOT_WAIT_FOR_NETWORK=1
CONFIG_CHECK_DIETPI_UPDATES=1
CONFIG_NTP_MODE=4
CONFIG_WIFI_COUNTRY_CODE=GB
CONFIG_SERIAL_CONSOLE_ENABLE=0
CONFIG_SOUNDCARD=none
CONFIG_LCDPANEL=none
CONFIG_PREFER_IPVERSION=ipv4
CONFIG_PREFER_IPVERSION=ipv4
CONFIG_NTP_MIRROR=debian.pool.ntp.org

SOFTWARE_EMONHUB_APIKEY=
SOFTWARE_VNCSERVER_WIDTH=1280
SOFTWARE_VNCSERVER_HEIGHT=720
SOFTWARE_VNCSERVER_DEPTH=16
SOFTWARE_VNCSERVER_DISPLAY_INDEX=1
SOFTWARE_VNCSERVER_SHARE_DESKTOP=0
SOFTWARE_OWNCLOUD_NEXTCLOUD_USERNAME=admin
SOFTWARE_OWNCLOUD_DATADIR=/mnt/dietpi_userdata/owncloud_data
SOFTWARE_NEXTCLOUD_DATADIR=/mnt/dietpi_userdata/nextcloud_data
SOFTWARE_WIFI_HOTSPOT_SSID=DietPi-HotSpot
SOFTWARE_WIFI_HOTSPOT_KEY=dietpi
SOFTWARE_WIFI_HOTSPOT_CHANNEL=3

DEV_GITBRANCH=master
DEV_GITOWNER=Fourdee

devingfx, 2021-08-14:
Hello,

Thank you for your great job creating DietPi, it's a time saver for us! ^^

I have questions on the `Automation_Custom_Script.sh` / hosted custom script, I did'nt found so much doc on it...
_( I'll shortcut "`Automation_Custom_Script.sh` / hosted custom script" to **CS** below )_

- Can we use several `AUTO_SETUP_INSTALL_SOFTWARE_ID` AND CS ?
- When does it start relative to SOFTWARE_ID ? before / after / parallel (like: Can I choose `Git` to be autoinstalled and use `git` in the CS ?
- Is script executed by `bash` ? Does it need hashbang (`#!/bin/bash`) ?
- What is the PWD of the script ? The user is `root ` ?
- Can it be used to craft configurations of packages installed via  `AUTO_SETUP_INSTALL_SOFTWARE_ID` ?

- Do you plan / already have some working exemples of CS ? ( local / online )

Thx very much if someone have time to answer !

Joulinar, 2021-08-14:
`AUTO_SETUP_INSTALL_SOFTWARE_ID` is used inside `dietpi.txt` on initial first run/setup only to install software title according their ID. It is not used later on or on other scripts

MichaIng, 2021-08-14:
There are actually two possible scripts for automation:
- `Automation_Custom_PreScript.sh` is executed pretty early on first boot, before the network is brought up, so it can also be used to do advanced network setups and such.
- Then comes first login, update, and `AUTO_SETUP_INSTALL_SOFTWARE_ID` installs are done.
- `Automation_Custom_Script.sh` runs after the software installs, so you can use it to do software setups/configs.
- I just tested to execute a script without shebang, and at least on Bullseye, `bash` executes those with `.sh` or `.bash` ending automatically with bash. Interesting find, however I'd always use a shebang regardless 😉. It's btw executed like that, so you may simply test it on another system to be sure it works as expected:
    ```sh
    /root/AUTO_CustomScript.sh 2>&1 | tee /var/tmp/dietpi/logs/dietpi-automation_custom_script.log
    ```
    It's moved to that location/name first, not sure why.
- The working directory remains unchanged, hence is the one of the calling script, which is in cases of our scripts a dedicated temporary directory like `/tmp/DietPi-Software`. This is automatically removed once the scripts exits, so can be used to create temporary files and don't need to cleanup.
- I guess there are examples floating around here on GitHub or in the forum, but I cannot find one on a quick search.

devingfx, 2021-08-14:
Great thank for your usefull answers @MichaIng !  
And such quickly ... ❤️ 

I'm trying to make a [odoo](odoo.com) box with it's community (GPL) version... I may look at adding it in DietPI Sotfware-List with tuto, but I won't guaranty when ^^

see ya~