MichaIng/DietPi #144: DietPi OrangePi PC Serial

scargill, 2015-12-14:
Hi

I am running Node-Red and have installed DietPi for the OrangePi PC having failed to get everything to work with previous installations of Debian.... but one thing I did get working at an earlier time was the Serial.

However, with DietPi no matter what I do I cannot get the SerialPort to appear in NodeRed - this is a dealbreaker for me as I use a Nextion Serial Display. 

I tried npm install serialport and that broke due to an early version of Node - I upgraded Node and the installation LOOKS ok - but no sign of the SerialPort.

For the OrangePi PC to be a viable alternative to the Raspberry Pi, surely it must be essential that the Ports and Serial work...

DietPi looks great but for the orange pi pc could you please look into making at least the serial port available.

Fourdee, 2015-12-14:
Hi Peter,

Many thanks for the feedback and report.

I'd like to check your system settings, please can you run the following on your device:
```
dietpi-bugreport
```
I'll need the reference CODE: once its completed.

ttyS0 is enabled on the OrangePi PC image:
[image]

Unfortunately, i do not currently own any serial adapters, so i'am unable to test your setup. However, as DietPi should support this, i've purchased a USB TO RS232 TTL PL2303 adapter and should be here in a few days. 
This will allow me to see if the issue is related to the DietPi image itself, or, NodeRed.

scargill, 2015-12-14:
Hi

 

Well it is easy to test the serial on the Pi – just hook input and output together – my point is that npn install serialport simply does not result in a serial port appearing in node-red – so there is no way for me to actually use the serial port.  

 

 <https://cloud.githubusercontent.com/assets/7077026/11782339/664a2fec-a268-11e5-89b0-1b482f767209.png>

Fourdee, 2015-12-14:
> Well it is easy to test the serial on the Pi – just hook input and output together – 

I'am not going to lie, my experience with GPIO/Hardware related software/projects is extremely limited. You'll have to excuse my unintentional ignorance.

> my point is that npn install serialport simply does not result in a serial port appearing in node-red – so there is no way for me to actually use the serial port.
This I can work with. 
Once the device arrives, i'll try to duplicate your installation and see if I can find the cause. As this is a new area for me, you will need to bear with me, once I get my bearings, i should be fine.

If you have the time, would you be able to post/link the steps you are taking to install and configure Node-red and your serial adapter, this should help speed things up.

scargill, 2015-12-15:
Absolutely.... it's worth it to make this work.

So I use your program to install NGINX/SQLITE/PHP - I would normally then put the latest Mosquitto (with websockets) on there but I missed that.

So the first thing is NPM and NODE

d /root
curl --progress-bar --location https://deb.nodesource.com/setup_0.12 |  bash -
apt-get -qq --yes -o=Dpkg::Use-Pty=0 --force-yes install nodejs
apt-get -qq --yes -o=Dpkg::Use-Pty=0 --force-yes install build-essential
apt-get -qq --yes -o=Dpkg::Use-Pty=0 --force-yes update

For some reason it's not putting the latest versions of these... that needs slight work (this is all from a script I use to setup Raspberry Pi - which I will summarily dump if your setup is better).

Then Node-Red

cd /root
npm cache clean
npm install bcrypt
npm install bcryptjs
npm install -g --unsafe-perm node-red
chmod 755 /etc/init.d/node-red
update-rc.d node-red defaults

I had to add in this "legacy" line which I've never needed before on a Raspberry Pi and then some modules including my own...

cd /root
apt-get install nodejs-legacy
mkdir .node-red
cd .node-red
npm install suncalc
npm install node-red-contrib-duskdawn
npm install node-red-contrib-scheduler
npm install node-red-contrib-grove
npm install node-red-contrib-thermostat
npm install node-red-contrib-bigtimer
npm install node-red-contrib-esplogin
npm install node-red-node-pushbullet
npm install node-red-contrib-freeboard
npm install node-red-node-openweathermap
npm install node-red-node-google
npm install node-red-node-sqlite
npm install moment

npm install node-red-contrib-ui

So this has set up Node-Red which can be accessed on xxx.xxx.xxx.xxx:1880

Normally a Raspberry Pi at this point would have a serial port NODE in Node-Red... however.

http://flows.nodered.org/node/node-red-node-serialport
(that comes from here http://flows.nodered.org/node/node-red-node-serialport

So that SHOULD be the serial port organised and in Node-Red you just need to tell it which one it is - I think in the case of the Orange Pi it is ttyS0 - though the one and only time I got this working in a manual installation on the Orange Pi it turned out to be ttyS3 - and it worked - here in this installation - nothing - zilch.

There is more for Node-Red for security but to keep it simple I'll not include that here.

Now, having in the setup told the Orange Pi to fire sound out of the 3.5mm jack, I install these

npm install -g normit
apt-get install mpg123

That should install an MP3 play routine.. and with normit installed at the command line you should be able to say

normit en en hello -t

You see "hello" appear after a couple of seconds (en en means the language - it uses Google translate) and -t means send an mp3 file of the returned word "Hello" to the mp3 routine - in other words out of the loudspeaker you should hear "hello". - instead -  nothing.

Finally - the ports... I got no-where with this - on a Raspberry Pi you get to control the ports from Node-Red - here I don't even know where to start - the Orange Pi website it pants and the only reference to ports was hidden behind a password and I don't think even exists now.  This board is THAT far off being a Raspberry-Pi beater, let down SOLELY by software.

Anything you can do to change that I'm sure will be WELL appreciated.

Any more - just shout...

scargill, 2015-12-15:
Ok here's the bug report reference which the OrangePiPC sent off to you.. 1e736aa11828-0

Fourdee, 2015-12-16:
Thanks Pete, appreciate the details.

Well my PL2303 adapter arrived today:
[image]

BSOD:
DRIVER_IRQL_NOT_LESS_OR_EQUAL
Been a while since i've seen one of those.

Would I be right in saying my PL2303 is faulty?

scargill, 2015-12-16:
Question – Orange Pi PC….+ DietPI PC+  Kodi – does that work yet?  If nothing else it would be worthwhile shoving it behind the TV as a media centre.

 

Pete.

Fourdee, 2015-12-16:
Hi Pete,
Please see my post here: DietPi for Orange Pi PC http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=662&fromuid=185718 (from OrangePi)

Date: Wed, 16 Dec 2015 08:03:33 -0800
From: notifications@github.com
To: DietPi@noreply.github.com
CC: daniel_haze@hotmail.com
Subject: Re: [DietPi] DietPi OrangePi PC Serial (#144)

Question – Orange Pi PC….+ DietPI PC+  Kodi – does that work yet?  If nothing else it would be worthwhile shoving it behind the TV as a media centre.

 

Pete.

Fourdee, 2015-12-18:
@scargill Waiting for my 2nd PL2303 to arrive (the 1st being faulty), then i'll be able to crack on with debugging and testing Serial adapter with node-red.

weakiwi, 2016-01-15:
The same problem in the Raspberry Pi B+(1g).
I delete the last '#' in the /etc/inittab (if you don't do this,the dietpi will show you boot message with login prompt)and I successfully login with v95.
But when I turn to the v105, it just show  'Uncompressing Linux... done, booting the kernel. ' (without any other message,just this sentence) with a pl2302-usb adapter but everything seems ok when using  monitor instead.

p.s:you must attach the gnd with the black pin and it will not shou

Fourdee, 2016-07-23:
Outdated, marking as closed. Please reopen if required.

closed 2016-07-23 by Fourdee