MichaIng/DietPi #146: apt-get error

dietcoke101, 2015-12-15:
I'm getting this apt-get error:

Apt-get has failed:
 - Apt string = php5-fpm 
 - Error code = 100   
 - Software Title = LEMP: Nginx / MySql / (PHP)

Was installing the LEMP after a new instance. 
Here is the last bit of the dpkg output:
<pre>
Creating config file /etc/php5/mods-available/readline.ini with new version
php5_invoke: Enable module readline for fpm SAPI
php5_invoke: Enable module readline for cli SAPI
Processing triggers for libc-bin (2.19-18+deb8u1) ...
Processing triggers for systemd (215-17+deb8u2) ...
Processing triggers for php5-fpm (5.6.14+dfsg-0+deb8u1) ...
Errors were encountered while processing:
 nginx-full
 nginx
E: Sub-process /usr/bin/dpkg returned an error code (1)
</pre>

and before that I had another error with something along these lines:
<pre>
Apt string = nginx xml-core --no-install-recommends
</pre>
<pre>
Preparing to unpack .../tcpd_7.6.q-25_armhf.deb ...
Unpacking tcpd (7.6.q-25) ...
Processing triggers for systemd (215-17+deb8u2) ...
Setting up nginx-full (1.6.2-5) ...
Job for nginx.service failed. See 'systemctl status nginx.service' and 'journalctl -xn' for details.
invoke-rc.d: initscript nginx, action "start" failed.
dpkg: error processing package nginx-full (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of nginx:
 nginx depends on nginx-full (>= 1.6.2-5) | nginx-light (>= 1.6.2-5) | nginx-extras (>= 1.6.2-5); however:
  Package nginx-full is not configured yet.
  Package nginx-light is not installed.
  Package nginx-extras is not installed.
 nginx depends on nginx-full (<< 1.6.2-5.1~) | nginx-light (<< 1.6.2-5.1~) | nginx-extras (<< 1.6.2-5.1~); however:
  Package nginx-full is not configured yet.
  Package nginx-light is not installed.
  Package nginx-extras is not installed.

dpkg: error processing package nginx (--configure):
 dependency problems - leaving unconfigured

</pre>

Fourdee, 2015-12-15:
Hi DietCoke,

Thanks for the report.
I need to see your system configuration. Can you run the following to send a bug report please:
```
dietpi-bugreport
```
I just the reference CODE:, you can leave the text document blank.

Fourdee, 2015-12-18:
@dietcoke101 Any chance of you sending the bug report? I'd appreciate it.

Fourdee, 2015-12-22:
No response. Please reopen if required.

closed 2015-12-22 by Fourdee