MichaIng/DietPi #147: USB Drive Install | Move dphys-swapfile to USB.

Fourdee, 2015-12-17:
As title.

Fourdee, 2015-12-17:
Available for v104 on new installations:
Dedicated USB drive installations will also move the swapfile to the USB drive.

closed 2015-12-17 by Fourdee

dschaper, 2015-12-18:
Just a small addition to this, the new version of PiHole has been updated and no longer requires a swapfile for memory considerations. As the ad-list domains are pulled, they are stored in the `tmp` directory instead of in memory.

Fourdee, 2015-12-18:
@dschaper DietPi mounts `/tmp` to ram (tmpfs) as a performance tweak. So the swapfile increase is still necessary for <=512mb systems.

dschaper, 2015-12-18:
Okay, good deal. I just noticed that the Pi-Hole install script still had the swapfile increase routine still set up...